schreibfaul1/ESP32-MiniWebRadio #612: Error OTA-upload in Version 3.6.1i  - Jan 17/2025

HWH53, 2025-01-25:
Hallo Wolle,

ich bin heute von Windows 10 auf Linux Mint Cinnamon 22.1 umgestiegen. Nach erfolgreicher Installation von VSCode habe ich das Repository des Miniwebradios geklont. Funktionierte ohne Fehler. Nachdem ich die erforderlichen Codeanpassungen durchgeführt und und fehlerfrei kompiliert hatte, konnte ich den Code nicht mehr über OTA uploaden. Siehe Auszug aus dem Protokoll. Ein upload ohne OTA funktioniert einwandfrei. 
Ich sehe das die Host IP-Adresse auf 0.0.0.0 gesetzt ist. Hat sich dort in den "Plattformfiles" wieder einmal ein Fehler eingeschlichen? 
Eine Überprüfung auf der Windowsinstallation brachte nach dem Update das gleiche erfolglose Erbebnis  (vorher funktionierte diese Uploadform). 

`Building in release mode
Retrieving maximum program size .pio/build/esp32s3_OTA/firmware.elf
Checking size .pio/build/esp32s3_OTA/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  23.3% (used 76268 bytes from 327680 bytes)
Flash: [======    ]  64.3% (used 2488026 bytes from 3866624 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = espota
Uploading .pio/build/esp32s3_OTA/firmware.bin
18:06:32 [DEBUG]: Options: Namespace(esp_ip='MiniWebRadio.local', host_ip='0.0.0.0', esp_port=3232, host_port=55336, auth='', image='.pio/build/esp32s3_OTA/firmware.bin', spiffs=False, debug=True, progress=True, timeout=10)
18:06:32 [INFO]: Starting on 0.0.0.0:55336
18:06:32 [INFO]: Upload size: 2488384
Sending invitation to MiniWebRadio.local 
18:06:32 [INFO]: Waiting for device...
18:06:42 [ERROR]: No response from device
*** [upload] Error 1
============================================================================ [FAILED] Took 20.56 seconds ============================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32s3_OTA    FAILED    00:00:20.558
======================================================================= 1 failed, 0 succeeded in 00:00:20.558 =======================================================================
`

schreibfaul1, 2025-01-25:
Hallo Helmut,
hab auch Linux Mint am Laufen. Das ist eine gute Entscheidung, ist viel besser als Windows. Allerdings ist meine Installation schon alt, und ich weiß nicht mehr, was ich damals alles gemacht habe. 

OTA funktioniert bei mir:
[image]

Du kannst aus der Anwendungsverwaltung "Avahi Discovery" installieren, das sieht alle "bonjour" Dienste

[image]

Vielleicht liest jemand mit und kann sagen, welches Paket noch fehlt.

vG Wolle

HWH53, 2025-01-26:
Hallo Wolle,

ich bin mittlerweile einen Schritt weiter. Unter Linux funktioniert der OTA-Upload wieder, warum auch immer. Bei Linux liegt das Problem nun nur noch an der Firewall, die den OTA-Upload blockiert. Erlaube ich per Firewall den eingehenden Datenverkehr kann ich auch über OTA uploaden. Dies sollte aber nicht Sinn der Sache sein.
Kanns Du, oder jemand aus der Community, weiterhelfen wie meine Firewallregeln unter Linux Mint 22.1 für den OTA-Upload aussehen müssen? 

Danke für die Mühe im voraus.

Helmut

HWH53, 2025-01-27:
Hallo zusammen,
Problem gelöst.

[image]

Eine Regel anlegen, Eingang erlauben, Protokoll TCP, Zieladresse = Host (VsCode - Maschine)

Helmut

closed 2025-01-27 by HWH53