schreibfaul1/ESP32-MiniWebRadio #613: small pins of the national flag

valera39, 2025-01-25:
> Hello Dear Wolle!
> You can add small icons of the flag of the country where the radio is currently playing. The data can be taken from the playlist, where the country of the radio station is specified, and place the flag collection on the map. 
> Maybe this is possible already in version 3.6?
> 
> [image]
> 
> And in the future, on large displays, it will be possible to make a separate information tab, where you can place the parameters of the playing radio station and other data that are not relevant on the main screen 

 _Originally posted by @valera39 in [#591](https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/591#issuecomment-2600784673)_

valera39, 2025-01-29:
Since there is no reaction to my proposal, I think my proposal is not worthy of consideration

schreibfaul1, 2025-01-29:
The problem is the implementation.
There is a character set with the flags that can be used on PCs. This is what I did here:

[image]

I can't do that on the TFT display, I would need about ~200 symbols. We'll have to postpone that until later.

valera39, 2025-01-29:
Thanks, I'll have to buy the display you recommended.

schreibfaul1, 2025-01-29:
Better wait a little longer. These displays (with integrated ESP32-S3) have not been around for very long. There are sure to be many more suppliers on the market. And I'm not making much progress, it's all going to take some time.

valera39, 2025-01-30:
Okay, I trust you completely

schreibfaul1, 2025-02-01:
I found this website, it works without much effort:
https://www.welt-flaggen.de/herunterladen/bilder

and have realised this:

[image]

That's not nice, but there was just enough space.

This will come with the next update, for which the "flags" folder must be extracted to the SD card.

valera39, 2025-02-01:
That's great! I'll definitely try it.
I look forward to seeing your developments with the larger 800x400 display.

valera39, 2025-02-01:
Maybe to make it more logical to move the station and number icons up next to the flag? There's enough space there.
What do you think?

schreibfaul1, 2025-02-02:
swapped the symbols

[image]

closed 2025-02-05 by valera39