schreibfaul1/ESP32-MiniWebRadio #614: Another build error ESP32-S3  WROOM N16R8

Bigpriap, 2025-01-25:
I find this project fascinating, but unfortunately, I am currently stuck as I cannot compile without errors.

I am using the latest version of Visual Studio Code, version 1.9.6.4, with the PlatformIO extension, and I have followed all the steps in the guide. Since I am not a programmer, I initially thought I was doing something wrong. However, today I asked a programmer friend to try it on his laptop, setting up everything from scratch.
For the ESP32 Wroom1 (N16R8) variant, I am getting the same compilation error:

Archiving .pio\build\esp32s3\esp-idf\framework-arduinoespressif32\libframework-arduinoespressif32.a
The command line is too long.
Compiling .pio\build\esp32s3\hal\rmt_hal.c.o
Compiling .pio\build\esp32s3\hal\pcnt_hal.c.o
*** [.pio\build\esp32s3\esp-idf\framework-arduinoespressif32\libframework-arduinoespressif32.a] Error 1
[FAILED] Took 73.38 seconds

I also tried IDF 5.3.1, but the issue persists.

Can anyone help me, please?

rjshiffer, 2025-01-26:
I believe this is the same error reported in #609 and #611.  I also get this error in Env [esp32s3].  Env [esp32] and [esp32s3_OTA] are successful.

ldisco, 2025-01-26:
Hello!
I have the same error when compiling.
I did everything according to the instructions. I tried different methods to fix the error..
How else can I compile the project without errors?

[image]

schreibfaul1, 2025-01-26:
The error seems to occur in Windows and is said to be related to the Arduino Core V3.1.0 
Arduino 3.0.7 and 3.1.1 should work, can someone please test this?

ldisco, 2025-01-26:
I use it 3.0.7

[image]

need to install 3.1.1 ?

Bigpriap, 2025-01-26:
Arduino  3.1.1 and IDF 5.3.2.241224  Win 11  Version 10.0.22631 Build 22631not working for me.

Arne65, 2025-01-26:
Hello Wolle,

I changed the following

Platformio.ini

[env]
extra_scripts = pre:env-extra.py

;platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip ;Arduino core 3.1.0 / IDF 5.3.1
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip    ;Arduino core 3.1.0 / IDF 5.3.2

seems to be working.

Flash: [======    ]  64.4% (used 2489878 bytes from 3866624 bytes)
Building .pio\build\esp32s3\firmware.bin
esptool.py v4.8.5
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
================================================================================================ [SUCCESS] Took 486.56 seconds ================================================================================================
Environment    Status    Duration
-------------  --------  ------------
esp32s3        SUCCESS   00:08:06.565
================================================================================================= 1 succeeded in 00:08:06.565 ================================================================================================

Jason2866, 2025-01-26:
Hello, it is a Windows limitation. Path length limit is max 32767 Bytes. For bigger projects that's not enough. I have a fix included in pioarduino which is not yet in a release build.
The fix is included in the pioarduino branches `develop` and `devel/v3.2.x`
It can be used by changing in platformio.ini the platform setting.
Example
```ini
platform = https://github.com/pioarduino/platform-espressif32.git#develop
```
or
```ini
platform = https://github.com/pioarduino/platform-espressif32.git#devel/v3.2.x
```

Jason2866, 2025-01-26:
@schreibfaul1  There is an option in pioarduino to change sdkconfig settings for Arduino without the need to build the project as Arduino as an component of IDF. I call it Hybrid Compile. This mode is developed from me and just for such use cases you have. Adjusting "some" sdkconfig  settings for Arduino. See example https://github.com/pioarduino/platform-espressif32/blob/6a5e1fcf97c3a378b20c48853667b017f05fc942/examples/arduino-blink/platformio.ini#L48-L72
The setup is same as for Arduino, just specify the sdkconfig entrys which should differ to the default Arduino sdkconfig settings. For questions please don't hesitate to ask me at pioarduino Discord https://discord.gg/Nutz9crnZr

schreibfaul1, 2025-01-26:
Hi @Jason2866,
thanks for the clarification and explanation of the problem. I will change the platform to the recommended develop-branch. 

And I will soon be trying out the "Hybrid Compile".

best regards
Wolle

Jason2866, 2025-01-26:
@schreibfaul1 Viele Grüße aus dem Süden von Deutschland. Tolles Projekt dein Webradio!
Werde sicher eines meiner S3 Display Devboards damit "ausstatten".

schreibfaul1, 2025-01-26:
@Jason2866 
Ich habe ein Elecrow 5Inch Display zum testen. Videos kann man damit nicht abspielen, dafür ist es zu langsam.
Für dieses Projekt scheint es aber schnell genug zu sein. Die SPI Displays sind auch nicht schneller.

viele Grüße aus Berlin :-)

Bigpriap, 2025-01-26:
Thank you very much! I tried the current version of the code, and it no longer gives a compilation error, so I eagerly uploaded it to the ESP32-S3-WROOM-1 N16R8 (at least that’s what it says on it) using the default settings in platformio.ini and the necessary modifications in common.h. However, the ILI9341 display did not light up.

I connected BL to 3.3V, and the backlight turned on, but that’s all!

Assuming the code is running correctly but the SD card is not working, how long does it take for an error message to appear?

I tried checking the Serial Monitor, and it doesn't seem right; the board keeps resetting, and the memory settings don’t seem to be OK:

���ESP-ROM:esp32s3-20210327
Build: Mar 27 2021
rst:0xc (RTC_SW_CPU_RST), boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403782c4
  #0  0x403782c4 in esp_restart_noos at C:\Users\Katty\.platformio\packages\framework-espidf\components\esp_system\port\soc\esp32s3/system_internal.c:160

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x104c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xaec
load:0x403cb700,len:0x2ce0
entry 0x403c889c
E (921) esp_psram: SPI SRAM memory test fail. 1700/262144 writes failed, first @ 3C326860
E (921) cpu_start: External RAM failed memory test!

abort() was called at PC 0x40377830 on core 0

Backtrace: 0x40378361:0x3fceb220 0x4037fc05:0x3fceb240 0x40387b5e:0x3fceb260 0x40377830:0x3fceb2d0 0x403cc9fd:0x3fceb340 0x403ccd19:0x3fceb380 0x403c8905:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
  #0  0x40378361 in panic_abort at C:\Users\Katty\.platformio\packages\framework-espidf\components\esp_system/panic.c:454
  #1  0x4037fc05 in esp_system_abort at C:\Users\Katty\.platformio\packages\framework-espidf\components\esp_system\port/esp_system_chip.c:92
  #2  0x40387b5e in abort at C:\Users\Katty\.platformio\packages\framework-espidf\components\newlib/abort.c:38
  #3  0x40377830 in call_start_cpu0 at C:\Users\Katty\.platformio\packages\framework-espidf\components\esp_system\port/cpu_start.c:587
Rebooting...

It seems like the board keeps rebooting due to an external RAM failure. Any ideas on how to fix this? Wrong ESP?

Jason2866, 2025-01-26:
@Bigpriap Looks like false PSRAM type. If you have an boards json used with QIO PSRAM use OPI PSRAM and vice versa. 
The ESP32-S3-WROOM-1 N16R8 i know uses QIO Flash and OPI PSRAM. 
In every case check your used board!