schreibfaul1/ESP32-MiniWebRadio #615: Fehlermeldung  i2c.master

skassel2024, 2025-01-28:
Hallo Wolle, 
in der aktuellen Version bekomme ich folgende Fehlermeldung:
Radio funktioniert ansonsten.

vG Sören
[image]

[image]

schreibfaul1, 2025-01-29:
Hallo Sören,
das kommt von den SDA/SCL Pins

[image]

Der Kommentar dahinter für kapazitive TP ist Quatsch. Die Pins möchte ich später zusätzlich für größere Displays mit dem Chip GT911 verwenden. Zur Zeit kann da nur ein Lichtsensor angeschlossen werden um die Displayhelligkeit zu dimmen. Wenn du die Pins nicht benutzt setze sie bitte auf -1.

vG
Wolle

skassel2024, 2025-01-29:
Danke, Problem gelöst :)