schreibfaul1/ESP32-MiniWebRadio #616: Not an issue, just a question.

Jefke1, 2025-01-29:
Hello everyone,

I would like to know which display to buy as there are so many different ones.
TN-ILI9488 or IPS-ILI9488.

https://www.aliexpress.com/item/1005001998578860.html?spm=a2g0o.order_list.order_list_main.378.1de679d2IG0gVQ&gatewayAdapt=glo2nld

Thank you in advance,
Greetings Jeffke1.

schreibfaul1, 2025-01-29:
Hello @Jefke1,
only displays with a touch panel are suitable. IlI9486 or IlI9488 does not matter.

Jefke1, 2025-01-29:
Hello Wolle,
Thank you for your quick response.
My biggest concern is 16 or 18bit color interface, otherwise the colors and logos will not be displayed correctly.
(0x55) or (0x66) ??

schreibfaul1, 2025-01-29:
RGB565 is used internally - 16 bits, 5 red, 6 green, 5 blue. The ILI9488 can do 24 bit but this is downsampled. You can use both. It looks like the shipping is more expensive than the display itself (the price shown is 0.99€) :-))

Jefke1, 2025-01-29:
really, mine says €7.82
thanks for the info.
Don’t stop exceeding our expectations, we couldn’t do it without you.
Greetings Jeffke1.

closed 2025-01-29 by Jefke1