schreibfaul1/ESP32-MiniWebRadio #617: menus will not working

mickeypop, 2025-01-31:
ref ESP32-S3-DevKitC-1-N16R8 with PCM5102A
all paths have been verified several times

NOTE the radio does play, though I cannot change stations etc...
the BH1750 does work but I still can't set thresholds.

in both the touch and web interfaces I can't see why they will not work

WEB is not loading any settings or showing any of the stations in the list though I does play the 42 stations in stations.json
 &nbsp;&nbsp;&nbsp;  if I use search to find a station it plays but overwrites my entire stations.json file

TOUCH if I touch the screen the speaker and 2 arrows show with 1 slider
  &nbsp;&nbsp; &nbsp;   if I touch the slider it just exits
  &nbsp;&nbsp;&nbsp;    if I touch any icon the other icon show but touching any icon in the 2nd screen and it exits

I should note there are no compiler errors

```
00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 29619.3 MB by 29664.0 MB free
[   283][W][main.cpp:449] defaultsettings(): 0x0000,  IR address
00:00:00 setup: ....  seek for WiFi networks
00:00:00 WiFI_info:   add SSID: mickeysXXXXXX
[  3148][W][STA.cpp:541] disconnect(): STA already disconnected.
00:00:00 WiFI_info:   Connecting WiFi...
00:00:00 WiFI_info:   mDNS name: MiniWebRadio
00:00:00 setup: ....  connected to mickeysXXXXX, IP address is 192.168.1.89, Received Signal Strength -32 dB
... station info removed ...
00:00:00 setup: ....  Ambient Light Sensor BH1750 found
....
10:20:46 click on ..  backpane
10:20:52 click on ..  backpane
10:20:54 click on ..  backpane
```

[image]
[image]

TubaFonse, 2025-02-01:
Hello mickeypop, 
your TOUCH issue, pls check your setup in common.h for the **Touchpanel-Rotation**,  is it **3** ? 
ILI9341 needs the following settings: 
#define TFT_CONTROLLER     0 
#define TP_VERSION              0 
#define TP_ROTATION            **3** 

maybe this helps ...

best regards, 
Alfons.

mickeypop, 2025-02-02:
i can after many boots get into the web interface but still no Touch
for REF

```
#define _SSID                   "mickeysXXXXX"                  // Your WiFi credentials here
#define _PW                     "XXXXXX"                // Or in textfile on SD-card
#define DECODER                 1                               // (1)MAX98357A PCM5102A CS4344... (2)AC101, (3)ES8388
#define TFT_CONTROLLER          0                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION       0                               // (0) off (1) on
#define TFT_ROTATION            3                               // 1 or 3 (landscape)
#define TFT_FREQUENCY           40000000                        // 80000000, 40000000, 27000000, 20000000, 10000000
#define TP_VERSION              0                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (6)ST7796RPI, (7)GT911
#define TP_ROTATION             3                               // 1 or 3 (landscape)
#define TP_H_MIRROR             0                               // (0) default, (1) mirror up <-> down
#define TP_V_MIRROR             0                               // (0) default, (1) mittor left <-> right
#define I2S_COMM_FMT            0                               // (0) MAX98357A PCM5102A CS4344, (1) LSBJ (Least Significant Bit Justified format) PT8211
#define SDMMC_FREQUENCY         80000000                        // 80000000 or 40000000 Hz
#define FTP_USERNAME            "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD            "esp32"
#define CONN_TIMEOUT            2500                            // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL        3500                            // encrypted connection timeout in ms (https://...)
#define WIFI_TX_POWER           10   
```

TubaFonse, 2025-02-02:
Hi Mickey, for **ILI9341**, its little tricky: 
I see, that you set **TFT**_ROTATION  = 3, then you must set **TP**_ROTATION = 1 , 
if you set TFT_ROTATION  = 1, then you must set TP_ROTATION = 3
please try it! 

Best regards, Alfons.

mickeypop, 2025-02-02:
TP_ROTATION=1 fixed it and all menus seem working except BT is grayed out
that's next issue

```
#define TFT_ROTATION            3                               // 1 or 3 (landscape)
#define TP_ROTATION             1                               // 1 or 3 (landscape)
```

closed 2025-02-03 by mickeypop