schreibfaul1/ESP32-MiniWebRadio #623: Bugs in the latest version.

hevet, 2025-02-05:
@schreibfaul1 Hi. I found a few errors, typos.

Lines in main.cpp file 2129, 2223, 2249, 2327 are .pngg file format instead of .png

Icon in btn/s - Button_Up_Magenta.png is too small, it is 30x30 and should be 40x40.

Lines 2423-2424 is
btn_RA_bt.show();
btn_RA_off.show(!_f_BtEmitterFound);
should be
btn_RA_bt.show(!_f_BtEmitterFound);
btn_RA_off.show();
otherwise after pressing ok on the remote control, the radio off icon will not appear in the icons menu.

hevet, 2025-02-05:
@schreibfaul1 :)

In main.cpp there is pic_BR_logo.setPicturePath("/common/Brightness.jpg"); which is why the background in the brightness menu is not displayed. On the card it is in .jpg.

The player does not refresh the number of the file being played, causing the numbers to overlap.

[image]

schreibfaul1, 2025-02-05:
I had previously written the audio file number /sum directly on the display, but that no longer works so well, I have now used a text box. And absolutely right, Brightness is a jpg file.
The layout of Brightness doesn't look good, I have no idea how to improve it. But I have lowered the slider.
You don't need to take a photo of the display, just click on the graphic in the info area.

[image]

valera39, 2025-02-05:
Sometimes the station flag icon disappears when changing the song being played. It reappears when you switch to a different station

hevet, 2025-02-06:
I confirm, the flag is not displayed when we return to the clock and sleep timer settings.

schreibfaul1, 2025-02-06:
I hope the problem has now been solved.

hevet, 2025-02-12:
@schreibfaul1 Hi. When dialing the station number from the remote control, the font got smaller, is this an intended effect or an error? 320x240 display.

schreibfaul1, 2025-02-12:
Thank you, I made a mistake. Now it's back the way it should be.

[image]

hevet, 2025-02-24:
@schreibfaul1 
In file main.cpp is 
`if(btnNr == 10){btn_EQ_balance.showClickedPic(); _toneBAL = 0; txt_EQ_balance.writeText( "0");  sdr_EQ_balance.setValue(_toneLP);  webSrv.send("settone=", setI2STone()); btn_EQ_balance.showAlternativePic();}
                            if(btnNr == 20){btn_EQ_lowPass.showClickedPic();  _toneLP = 0; txt_EQ_lowPass.writeText( "0");  sdr_EQ_lowPass.setValue(_toneLP);  webSrv.send("settone=", setI2STone()); btn_EQ_lowPass.showAlternativePic();}
                            if(btnNr == 30){btn_EQ_bandPass.showClickedPic(); _toneBP = 0; txt_EQ_bandPass.writeText("0"); sdr_EQ_bandPass.setValue(_toneLP); webSrv.send("settone=", setI2STone()); btn_EQ_bandPass.showAlternativePic();}
                            if(btnNr == 40){btn_EQ_highPass.showClickedPic(); _toneHP = 0; txt_EQ_highPass.writeText("0"); sdr_EQ_highPass.setValue(_toneLP); webSrv.send("settone=", setI2STone()); btn_EQ_highPass.showAlternativePic();}`
`if(btnNr == 20){btn_EQ_lowPass.showClickedPic();  _toneLP = 0; txt_EQ_lowPass.writeText( "0");  sdr_EQ_lowPass.setValue(_toneLP);  webSrv.send("settone=", setI2STone()); btn_EQ_lowPass.showAlternativePic();}
                            if(btnNr == 30){btn_EQ_bandPass.showClickedPic(); _toneBP = 0; txt_EQ_bandPass.writeText("0"); sdr_EQ_bandPass.setValue(_toneBP); webSrv.send("settone=", setI2STone()); btn_EQ_bandPass.showAlternativePic();}
                            if(btnNr == 40){btn_EQ_highPass.showClickedPic(); _toneHP = 0; txt_EQ_highPass.writeText("0"); sdr_EQ_highPass.setValue(_toneHP); webSrv.send("settone=", setI2STone()); btn_EQ_highPass.showAlternativePic();}`

mogart, 2025-02-25:
Hi,
Hello. I'm a beginner, but I really want to get the ESP32miniWbradio to work. I downloaded it from Github, unzipped it, and tried to compile platformio.ini with VSC, but I got an error and can't go any further. This error: "Bad CMake executable: "". Check to make sure it is installed or the value of the "cmake.cmakePath" setting contains the correct path". Please tell me what to do.
Does ESP32miniWebradio have a group on Facebook or other sites?

________________________________
差出人: Jacek ***@***.***>
送信日時: 2025年2月24日 17:11
宛先: schreibfaul1/ESP32-MiniWebRadio ***@***.***>
CC: Subscribed ***@***.***>
件名: Re: [schreibfaul1/ESP32-MiniWebRadio] Bugs in the latest version. (Issue #623)

[hevet]hevet left a comment (schreibfaul1/ESP32-MiniWebRadio#623)<https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/623#issuecomment-2677687442>

@schreibfaul1<https://github.com/schreibfaul1>
In file main.cpp is
if(btnNr == 10){btn_EQ_balance.showClickedPic(); _toneBAL = 0; txt_EQ_balance.writeText( "0"); sdr_EQ_balance.setValue(_toneLP); webSrv.send("settone=", setI2STone()); btn_EQ_balance.showAlternativePic();} if(btnNr == 20){btn_EQ_lowPass.showClickedPic(); _toneLP = 0; txt_EQ_lowPass.writeText( "0"); sdr_EQ_lowPass.setValue(_toneLP); webSrv.send("settone=", setI2STone()); btn_EQ_lowPass.showAlternativePic();} if(btnNr == 30){btn_EQ_bandPass.showClickedPic(); _toneBP = 0; txt_EQ_bandPass.writeText("0"); sdr_EQ_bandPass.setValue(_toneLP); webSrv.send("settone=", setI2STone()); btn_EQ_bandPass.showAlternativePic();} if(btnNr == 40){btn_EQ_highPass.showClickedPic(); _toneHP = 0; txt_EQ_highPass.writeText("0"); sdr_EQ_highPass.setValue(_toneLP); webSrv.send("settone=", setI2STone()); btn_EQ_highPass.showAlternativePic();}
if(btnNr == 20){btn_EQ_lowPass.showClickedPic(); _toneLP = 0; txt_EQ_lowPass.writeText( "0"); sdr_EQ_lowPass.setValue(_toneLP); webSrv.send("settone=", setI2STone()); btn_EQ_lowPass.showAlternativePic();} if(btnNr == 30){btn_EQ_bandPass.showClickedPic(); _toneBP = 0; txt_EQ_bandPass.writeText("0"); sdr_EQ_bandPass.setValue(_toneBP); webSrv.send("settone=", setI2STone()); btn_EQ_bandPass.showAlternativePic();} if(btnNr == 40){btn_EQ_highPass.showClickedPic(); _toneHP = 0; txt_EQ_highPass.writeText("0"); sdr_EQ_highPass.setValue(_toneHP); webSrv.send("settone=", setI2STone()); btn_EQ_highPass.showAlternativePic();}

valera39, 2025-02-25:
Hi [mogart](https://github.com/mogart)! Check if you have all the necessary VSC extensions installed.
I'm perfectly capable of handling these
[image]