schreibfaul1/ESP32-MiniWebRadio #625: Flags

skassel2024, 2025-02-06:
Hallo Wolle,
das neue Layout sieht gut aus. Bei den Flags ist mir aufgefallen, bei einigen deutschen Radiosendern wird als Länderkennung "D" oder "DE" mitgesendet. Im Verzeichnis Flags fehlt aber "d.jpg". Es werden also nur Flags für die Länderkennung "DE" angezeigt. 
Das nur als Hinweis. Durch kopieren und umbenennen von "de.jpg" in d.jpg funktioniert es bei allen deutschen Stationen.
Auch fehlt z.B. Flag für UK.
vG Sören

schreibfaul1, 2025-02-06:
Hi Sören,
danke für das Lob. Ich bin dabei den Hintergrund voll transparent zu gestalten. Beispielsweise ist bei der Zeit ist der Hintergrund noch schwarz.
Bei der Länderkennung nehme ich das was der Radiobrowser liefert. Das ist für Deutschland "DE" und für UK ist das "GB". Das passt zu den Namen der Flaggen, die habe ich nicht selbst erstellt sondern so aus den Internet kopiert.

beste Grüße
Wolle

skassel2024, 2025-02-07:
Hallo Wolle, hab das erst jetzt gesehen, Du nimmst die Länderinformation aus der stationlist und nicht aus dem stream. Daher muss nur in der stationlist die passende Länderkennung eingetragen werden. Das kopieren und umbenennen der Flags ist somit nich notwendig. sorry
vG Sören

schreibfaul1, 2025-02-07:
genau, ich habe mir das so vorgestellt:
1) in Radiobrowser eine Station auswählen

[image]
die Länderkennung wird automatisch übernommen, dann auf den Button mit dem Pfeil nach links

[image]
überträgt die Daten in die letzte Spalte der Tabelle,  du kannst mit der linken Maustaste dann den Sender in eine andere Spalte einsortieren, und dann erscheint auch die Flagge auf dem Display

[image]