schreibfaul1/ESP32-MiniWebRadio #626: IP Adresse wird nicht vollständig angezeigt

skassel2024, 2025-02-06:
Hallo Wolle,
bei der Anzeige der IP Adresse im Display werden 2 Stellen am Ende abgeschnitten.
vG Sören

schreibfaul1, 2025-02-06:
Ich habe die Icons etwas zusammengeschoben das wird jetzt okay sein. Die IP Adresse kann unterschiedlich lang sein, abhängig von der letzten Zahl (1 ...3 Stellen). Sie wird jetzt zentriert, bzw falls der Platz nicht reichen sollte wird die nächst kleinere Schriftgröße verwendet (was hoffentlich nicht nötig ist).
Und der Bildschirm ist jetzt vollständig transparent. Du kannst dein eigenes Hintergrundbild entwerfen oder von der KI zeichnen lassen :-) 

[image]

skassel2024, 2025-02-07:
danke funktioniert