schreibfaul1/ESP32-MiniWebRadio #627: Code Request - reset needed for WiFi reconnect

mickeypop, 2025-02-07:
code request

A minor bug I noticed

My WiFi gets bumped occasionally for a few seconds and the LCD changes to the disconnect status and only a reset reconnects.

An "if then else" approach in loop() fixes this and makes for an auto-reconnect.

I have used this example structure for about 10 years with good luck.
Though written for Arduino with adaptation it should work well.
```
void loop()
{
  if ( WiFi.status() ==  WL_CONNECTED )      // Main connected loop
  { 
	// ANY MAIN LOOP CODE HERE

  }  	// END Main connected loop()
  else
  {  	// WiFi DOWN - write wifi down status... reconnecting to LCD

     WFstatus = getWifiStatus( WFstatus );   // get current state

     WiFi.begin( SSID , Password );		// reconnect
     int WLcount = 0;
     while (WiFi.status() != WL_CONNECTED && WLcount < 200 )  	// while, could be rewriten for LCD display
     {
      delay( 100 );
         Serial.printf(".");
         if (UpCount >= 60)  // keep from scrolling sideways forever
         {
            UpCount = 0;
               Serial.printf("\n");
         }
         ++UpCount;
         ++WLcount;
     }
    if( getWifiStatus( WFstatus ) == 3 )   //wifi returns
    { 
		// clear WiFi down status
    }
     delay( 500 );
  }  // END WiFi down 
} // END loop()
```

schreibfaul1, 2025-02-10:
Thanks for your contribution, I will test this in the next few days.

mickeypop, 2025-02-10:
I hope this work
when designing for a Startup we discovered in a power failure our device booted up before the router and we never reconnected until a reset
this was our fix and I have always used it since