schreibfaul1/ESP32-MiniWebRadio #628: AI Thinker A1S support

OK1FWG, 2025-02-09:
Hi,
with the latest downloaded version I am not able to compile the code for my AI Thinker A1S board [esp32]. As I have more of those boards, I would like to build a radio  on it. Few weeks ago it was flawlessly compiled, but something changed. Thanks.

schreibfaul1, 2025-02-10:
Hi @OK1FWG ,
I encountered an error during the last updates that prevented compilation with the ESP32. The error did not show up with the ESP32-S3, so I had not noticed it. I have just updated my AI Thinker Radio myself and it is working again.

[image]

OK1FWG, 2025-02-10:
Perfect, will try to compile asap. Have to connect some kind of display. I have build about three V2 radios in the past. I really appreciate your work. I am glad for the AI Thinker board support, although it doesn't have enough I/O ports.