schreibfaul1/ESP32-MiniWebRadio #629: Change of the color for Station name, time , mode name

JozefKc, 2025-02-11:
Hello Wolle,
many thanks for the work you do. I would like to ask if it would be possible to simplify the way /e.g. with a shortcut in common.h/ to change the colors of the station name, subtitles, time and IP address. It would certainly be welcomed by many who do not like the current colors.
Thanks in advance
Josef

schreibfaul1, 2025-02-12:
That's an idea for later. The colours are fixed. You can change them here:

[image]

JozefKc, 2025-03-03:
Hallo Wolle!
Danke fuer die Unterstuetzung. Jetzt habe ich die Farben gaendert,

[image]