schreibfaul1/ESP32-MiniWebRadio #630: Avoid Duplicate Entries in saved BT MAC address and BT Name

TubaFonse, 2025-02-12:
Hallo Wolle, 
kleine Korrektur bei Gelegenheit, falls Aufwand vertretbar:  

1.) Doppelt gespeicherte BT MAC Adressen +BT Name sollten wieder manuell gelöscht bzw. überschrieben werden können - bzw. Doppeleinträge möglichst schon beim Speichern verhindert werden. 
Derzeit kann nur die komplette Liste **aller** Einträge mit "Clear" gelöscht werden und muss dann wieder mit den gewünschten BT-Geräten neu angelegt werden. 

[image]

2.) In der eingeblendeten Kommentar-Box bei "Clear" versteckt sich noch ein Tippfehler: 
  **clea_t_** all saved items => clea**r** all saved items: 

[image]

zu 1.) noch eine Frage: wo werden diese BT-Einstellungen gespeichert? Sind diese Einträge direkt im KCX-BT-Modul permanent abgelegt? Könnten diese Einträge auf der SD-Karte zusätzlich gesichert, editiert und neu geladen werden? 

Danke und viele Grüße, Alfons.

schreibfaul1, 2025-02-14:
Hallo Alfons,
wenn es keine Einträge in der Liste gibt verbindet sich der BT-Emitter mit dem ersten gefundenen Gerät. Wenn du mehrere BT Geräte hast kann das ungünstig sein.
Sind Einträge vorhanden verbindet sich der BT-Emitter nur noch mit diesen Geräten, alle anderen werden ignoriert. Die Liste kann nur komplett ausgelesen werden, und es gibt nur ein Kommando um die komplette Liste zu löschen.

Anscheinend reicht es nicht auf der Webseite doppelte Einträge zu blockieren, das muss ich wahrscheinlich direkt in der BT Library machen.  Das wird noch etwas dauern. Bis dahin... vG Wolle

TubaFonse, 2025-02-15:
Hallo Wolle, 
vielen Dank - gut zu wissen, dass sich der BT-Emitter immer mit dem ersten gefundenen Gerät verbindet, solange die Liste keine Einträge enthält. Damit kann ich sehr gut leben! 
Dann stören auch versehentliche Doppeleinträge nicht, da reicht mir das "Clear" völlig, um dann die komplette Liste zu löschen. 
**Also: Hier keine Änderung nötig -  denn Deine Zeit ist viel zu kostbar - viel wichtiger sind die nächsten Features für die nächsten Versionen des MWR!** Herzlichen Dank & Grüße, Alfons.

closed 2025-02-15 by TubaFonse