schreibfaul1/ESP32-MiniWebRadio #634: ESP32 S3 N16R8 CAM -  A better module for the project ?

slot00753, 2025-02-16:
Dear Wolle!
First, Thanks very much for delivering this amazing project to us diy audiophiles.

Playing around it for a while, I have had much pain with micro-SD storage (even at speed of 10MHz). 
I thought it would be nice to have it on the module - no wires, no bad contacts, no failed mounts.
Ealier I spent some time with ESP32 CAM which had a micro SD holder onboard.
So why not use its successor - ESP32-S3 N16R8 CAM module?
Say, this one from aliexpress. Search for "esp32 n16r8 cam"

[image]

[image]

I obtained one - and read/write/mount errors have gone! Raised speed to 80 MHz - ok.
Migration from ordinary S3 went seemless, no software updates needed except redefining
 
#define SD_MMC_D0  40 //Please do not modify it.
#define SD_MMC_CLK 39 //Please do not modify it. 
#define SD_MMC_CMD 38 //Please do not modify it.

Sorry for my noise if many of you have already found this device and use it.

schreibfaul1, 2025-02-16:
I don't know this board. I think it will work.
You can also connect the SD card directly (3.3V, GND, CMD, D0, SCK) you don't need any other components.

[image]

The current SD cards all manage 80MHz.

slot00753, 2025-02-16:
The bottom side of the above module with onboard microSD holder:

[image]

schreibfaul1, 2025-02-17:
You can unpack the data onto the SD card and compile and run. You can then see what happens in the terminal. If everything is OK, the programme will run without the DAC and display and you will have access via the web browser.

slot00753, 2025-02-17:
Dear Wolle, I am afraid you did not get me right.
Due to your efforts I've  built a fully functional MiniWebRadio project (except OTA :() on esp32s3 but with unreliable and faulty micoSD storage unit. I tried all possible options but was not satisfied until I found esp32s3 n16r8 cam module with onboard microSD holder which I mentioned in the beginning of my post.

I've just wondered if my diy-colleagues know about it  (or I am the only one who don't).  No doubt I recommend it for this project

closed 2025-04-03 by github-actions[bot]