schreibfaul1/ESP32-MiniWebRadio #635: meaning of release announcement "This is the last version with the ESP32 !!"?

mi-hol, 2025-02-19:
release announce states "This is the last version with the ESP32 !!"

should this mean "latest" or "final" = end of development for ESP32?

Please clarify

hevet, 2025-02-19:
Yes last one for esp32, esp32 support was removed from test version 4.0.

schreibfaul1, 2025-02-19:
The ESP32 is obsolete. PSRAM is used extensively in this project. The PSRAM frequency is limited to 40MHz with the ESP32, with the ESP32-S3 it is 120MHz.
Version 4 adds RGB displays (800x400 pixels), which is not possible with the ESP32.
In addition, the ESP32-S3 is equivalent in terms of availability and price. There is no need to continue supporting old ESP32.

Kicadder, 2025-02-19:
Nice! Any (little) idea of the availability of this (new, welcome, wonderful, wunderbar!) version. Several weeks, several months?
Is it something we can ferret about in advance (ucontroller, display modules, rotary encoder), just to be in the starting blocks when this version will appear? We are not wanting to garbage time, not any (micro-)seconds :-)

mi-hol, 2025-02-19:
> The ESP32 is obsolete.

the trouble with this statement is caused by the fact that "ESP32 " is generally used to identify a family of microcontrollers (aka category of MCUs) but in this context it is meant to identify the first model from the ESP32 series and there specifically the chip ID ESP32-D0WDR2-V3 . I'd highly recommend to revise the wording of release announcement in order to avoid confusing a large audience.

mi-hol, 2025-02-19:
instead of ""This is the last version with the ESP32 !!""
better ""This is the last version supporting the first model from ESP32 family with chip ID  ESP32-D0WDR2-V3!!"

schreibfaul1, 2025-02-19:
I think that an ESP32-S3 board will be easy to obtain, especially as it is available in large quantities on the usual online platforms. The ESP32 has been used in this project for about 5 years. It has now reached the limit of its capabilities. Every change to the software has to be tested on both µCs, which is simply too time-consuming.

mi-hol, 2025-02-23:
> It has now reached the limit of its capabilities. Every change to the software has to be tested on both µCs, which is simply too time-consuming.

@schreibfaul1 um das geht es doch gar nicht! Es geht nur um die total missverständliche Ausdrucksweise im englischen Announcement.

hevet, 2025-02-23:
Hmm, I don't know what's confusing here, I think you're being a bit picky.

valera39, 2025-02-23:
Absolutely don't see any problem here with the author's warning. I think that the user is not so stupid that he would not realize that it concerns discontinuation of ESP32 wrower support.

closed 2025-04-09 by github-actions[bot]