schreibfaul1/ESP32-MiniWebRadio #637: ESP32-S3-Touch-LCD-7 Waveshare

skassel2024, 2025-02-24:
Hi Wolle, bin begeistert über V4 pre-release. Ich habe das mit dem Waveshare Display gemacht. Funktioniert soweit alles. Nur brauche ich Deine Hilfe und einen Denkanstoss, wo und wie ich bei diesem Teil die I2S_DOUT, I2S_BCLK und I2S_LRC finde. Du hast dazu in common.h die Pins 19, 20, 15 definiert.
vG Sören

[image]

schreibfaul1, 2025-02-25:
Sehr gut!
Beim Waveshare Display sind alle GPIOs beschaltet. CAN-Bus, I2C-Portexpander und RS485 Schnittstelle werden nicht verwendet und können "tot" gelegt werden.
Es werden 3 GPIOs für den DAC benötigt, eventuell noch ein Pin für die Steuerung der Hintergrundbeleuchtung und ein Pin für den IR-Empfänger.
 
[image]

Die bestehenden Verbindungen müssen dazu getrennt werden. Am besten so, dass das Display ggf. wieder in den Originalzustand zurückgebaut werden kann.

skassel2024, 2025-02-25:
Danke Wolle,
funktioniert. GPIO15, GPIO 19 und GPIO20 vom ESP auf eine zusätzliche 5pol. Steckerleiste, Brücke H2 von 3V3 auf 5V und VCC/GND von i2C Stecher auf die zusätzliche5pol. Steckerleiste, von U6 pin 1 ablöten. Das wars :))
Das Projkt ist einach nur genial. Danke nochmals und viele Grüße
Sören

[image]

skassel2024, 2025-02-26:
Hallo Wolle,
noch eine Anmerkung: Bei der Auswahl TFT_FONT=3 (TFT_ARIAL), fehlen bei der Darstellung Stationsnummer und Timer jeweils die letzte Stelle. Siehe auch Bild ganz oben. Ich habe dann folgende Änderung vorgenommen:
In main.cpp Zeile 390 bei footerFontSize den Wert von 38 auf 34 geändert. Damit funktioniert es.
vG Sören

skassel2024, 2025-02-26:
Bin ein dankbarer Tester, weil's einfach Spass macht. Ich habe in den letzten Monaten einiges dazu gelernt :))
vG Sören

PalkoPinti, 2025-03-14:
Hallo Sören. Ich bin Laie und habe keine Programmierkenntnisse. Ich habe bereits ein MiniWebRadio aus der ESP32 WROWER-Platine erstellt und es funktioniert gut. Ich habe auch dieselbe Waveshare ESP32S3 Touch LCD 7-Platine, weiß aber nicht, wie ich V4 darauf bekomme. Das habe ich bereits versucht, aber das Display lässt sich überhaupt nicht einschalten. Könnten Sie mir helfen?

skassel2024, 2025-03-14:
Hallo PalkoPinti,
ich verwende VS Studio Code mit platformio, um das Programm aufzuspielen. Verwende dazu das How to install.pdf von Wolle. Wähle beim clonen nicht master sondern V4-pre-release. Alternativ download der zip Datei, entpacken und den entpackten Ordner in platformio zu Arbeitsbereich hinzufügen. "Content_on_SD_Card.zip" auf microSD entpacken. Ohne SD geht nix.
Jetzt noch in der common.h in Zeile 9 #define TFT_CONTROLLER 7 für RGB display und ab Zeile 91 die richtigen Werte für das Waveshare Display eintragen. Diese findest Du unter docs\rgb displays\Waveshare 7' Display\Waveshare_7inch.txt. Vergiss nicht entweder in common.h Zeile 7 und 8 Deine Wifi Zugangsdaten oder in der Datei networks.csv einzutragen. Dann noch etwas Lötarbeit, wie weiter oben beschrieben, um das PCM5102 Modul anzuschliessen.
viel Erfolg 
Gruß Sören

PalkoPinti, 2025-03-18:
Vielen Dank für Ihre Antwort. Ich habe es geschafft, es zum Laufen zu bringen. Ich habe eine andere Frage. Könnte man nicht anstelle des KCX_Emiter das eingebaute BT im ESP32 verwenden?

PalkoPinti, 2025-03-18:
[image]
Ich muss noch das DAC-Modul anschließen, das bestellt ist und noch nicht angekommen ist.

schreibfaul1, 2025-03-18:
Sieht gut aus, Gratulation!
Der ESP32-S3 kann nur BLE, das ist ein Bluetooth Modus um Geräte zu steuern aber nicht um Audio zu übertragen.

viele Grüße
Wolle

closed 2025-05-02 by github-actions[bot]