schreibfaul1/ESP32-MiniWebRadio #638: Display touch doesn't work properly

linoreale, 2025-02-27:
Hi everyone, 
I built the miniweb radio based on your project and it works, but there's a problem with display's touch system.
I'm using a ESP32-S3 N16R8 board and an ILI9488 display. On system boot it shows esp web radio logo and then last radio starts playng, but touch screen doesn't respond correctly, if I touch an icon, the led on esp board blinks but no action happens or wrong action or function starts.
Could it be a display problem, a wrong wiring or a known software problem?
Thanks and regards

Michele

skassel2024, 2025-02-27:
richtige TP_VERSION, TP_ROTATION, TP_H_MIRROR, TP_V_MIRROR in common.h ?
Bei ILI9488 TP_VERSION auf 4 setzen. Standard ist 5 (ST7796)

[image]

linoreale, 2025-02-27:
Hi skassel2024, 
yes, I set 4 for ILI9488 and left other as default. 
What can I try  in addition?
Thanks and regards

schreibfaul1, 2025-02-27:
Some displays have a design flaw, maybe it's the same with your display?
https://github.com/schreibfaul1/ESP32-MiniWebRadio/tree/V4-pre-release?tab=readme-ov-file#display

valera39, 2025-02-27:
[linoreale](https://github.com/linoreale), Try installing the touchpad version from a different display, such as ILI9486

linoreale, 2025-02-27:
Okay, 
so, have I try to disconnect TFT_DC or TFT_CS pin?
Thanks a lot to you all!
Regards

valera39, 2025-02-27:
[linoreale](https://github.com/linoreale), TFT_DC and TFT_CS do not need to be disabled

linoreale, 2025-03-01:
Hi,
on common.h I modified all the availlable variables but it's still not working. Screen shows images correctly but it seems touch pad is rotated on 90 degrees and no option modfies this.
Tried setting different display, rotation mirror and so on but it doesn't woek
what can I try again?
above my commn.h 
```
// clang-format off
#define _SSID                   "050153"                        // Your WiFi credentials here
#define _PW                     "amorredimicionne"                // Or in textfile on SD-card
#define DECODER                 1                               // (1)MAX98357A PCM5102A CS4344... (2)AC101, (3)ES8388
#define TFT_CONTROLLER          4                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION       0                               // (0) off (1) on
#define TFT_ROTATION            1                               // 1 or 3 (landscape)
#define TFT_FREQUENCY           40000000                        // 80000000, 40000000, 27000000, 20000000, 10000000
#define TP_VERSION              6                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (6)ST7796RPI, (7)GT911
#define TP_ROTATION             1                               // 1 or 3 (landscape)
#define TP_H_MIRROR             0                               // (0) default, (1) mirror up <-> down
#define TP_V_MIRROR             0                               // (0) default, (1) mittor left <-> right
#define I2S_COMM_FMT            0                               // (0) MAX98357A PCM5102A CS4344, (1) LSBJ (Least Significant Bit Justified format) PT8211
#define SDMMC_FREQUENCY         80000000                        // 80000000 or 40000000 Hz
#define FTP_USERNAME            "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD            "esp32"
#define CONN_TIMEOUT            2500                            // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL        3500                            // encrypted connection timeout in ms (https://...)
#define WIFI_TX_POWER           5                               // 2 ... 21 (dBm) Adjust the WiFi transmission power to optimise power consumption or increase range, default: 5

```

valera39, 2025-03-01:
[linoreale](https://github.com/linoreale), and have you tried this combination TFT_ROTATION 1 and define TP_ROTATION 3
or TFT_ROTATION 3 and define TP_ROTATION 1 ?

valera39, 2025-03-01:
Do you have the same display?
[image]

linoreale, 2025-03-01:
Hi valera39,
It seems the same.
Anyway, have I change values in common.h , build , compile and upload each time?
Thanks a lot, I'll try you suggestion.
Regards

valera39, 2025-03-01:
[linoreale](https://github.com/linoreale), Yes, exactly like that, change one parameter, compile, test. If it doesn't fit, change it again and so on until we get what we need.
I also fiddled with such a display for a long time until I set it up correctly :)
I don't buy such displays anymore

valera39, 2025-03-01:
[linoreale](https://github.com/linoreale).  Here I found in my archive common.h from the second version of the radio with such display. Try it, maybe it will help
#define _SSID               "***"                        // Your WiFi credentials here
#define _PW                 "***"
#define DECODER             0                               // (0)VS1053 , (1)MAX98357A PCM5102A... (2)AC101 (3)ES8388 (4)WM8978
#define TFT_CONTROLLER      4                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION   0                               // (0) off (1) on
#define TFT_ROTATION        1                               // 1 or 3 (landscape)
#define TFT_FREQUENCY       40000000                        // 27000000, 40000000, 80000000
#define TP_VERSION          3                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (3)ST7796RPI
#define TP_ROTATION         3                               // 1 or 3 (landscape)
#define AUDIOTASK_CORE      0                               // 0 or 1
#define AUDIOTASK_PRIO      2                               // 0 ... 24  Priority of the Task (0...configMAX_PRIORITIES -1)
#define SDMMC_FREQUENCY     20000000                        // 40000000, 2000000, 10000000, not every SD Card will run at 40MHz
#define FTP_USERNAME        "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD        "esp32"
#define CONN_TIMEOUT        500                             // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL    2000                            // encrypted connection timeout in ms (https://...)

linoreale, 2025-03-06:
Hi, 
I tried all availlable displays in the code list but touch panel doesn't still work properly, it remains 90° rotated, even though I set rotation 1 or 3 on each display version.
What else can I try? Is there something else to change in the code?
What display version works for sure, so I buy it?
Thanks and regards

linoreale, 2025-03-10:
Hi,
I'm wondering if there are any other tp_rotation option to use in addition to provided options or if it's possible to use other display drivers in the code
thamks

linoreale, 2025-03-16:
Hi, 
are there any other options for tp_rotation in addition to 1 and 3?

valera39, 2025-03-16:
Hi,
I set it to 2, but I got portrait mode. So there are no other options

valera39, 2025-03-16:
This kind of display has kind of a mirrored touchpad. You could of course ask Wolle to add changes to the code for such displays, but I think he is very busy with the new version or other things

linoreale, 2025-03-16:
Valera39, please suggest me a display that works for sure on this project, so I can buy a new one.
Thanks

valera39, 2025-03-16:
[linoreale](https://github.com/linoreale), I can't show a photo as the displays are already built into the radio, but bought it here, works great https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/649#issuecomment-2716946401

closed 2025-05-01 by github-actions[bot]