schreibfaul1/ESP32-MiniWebRadio #639: GT911 touch 无法使用

Big-mouth-c, 2025-02-28:
我测试您的项目，使用了GT911 touch IC。但触摸屏偶尔才能工作。
我查看的 函数
 void TP_GT911::reset() {
    vTaskDelay(10 / portTICK_PERIOD_MS);
    pinMode(m_intPin, OUTPUT); digitalWrite(m_intPin, LOW);
    pinMode(m_rstPin, OUTPUT); digitalWrite(m_rstPin, LOW);
    vTaskDelay(10 / portTICK_PERIOD_MS);
    digitalWrite(m_intPin, m_addr);
    pinMode(m_rstPin, INPUT);
    vTaskDelay(10 / portTICK_PERIOD_MS);
    digitalWrite(m_intPin, LOW);
    vTaskDelay(10 / portTICK_PERIOD_MS);
}
不知道  digitalWrite(m_intPin, m_addr); 这个语句的作用，是否只是个手误？

schreibfaul1, 2025-02-28:
Thank you for your contribution,
the GT911 routines are not an in-house development, but were taken from the Arduino GT911 standard library. I made a mistake in the process.
` digitalWrite(m_intPin, m_addr);` is rubbish.

The GT911 can recognise the number of fingers on the display and gestures.
In this project, it is sufficient if the GT911 is queried cyclically and returns the X/Y position.
The RESET and INTERRUPT pins are not used.

Big-mouth-c, 2025-02-28:
感谢您的回复。
因为设备的问题 TwoWire(0)被ES8388、xl9555占用，GT911我使用了 TwoWire(1)。
我再重新测试代码，将GT911设置为TwoWire(0)。
顺便说一下您的程序非常好！！！

closed 2025-02-28 by Big-mouth-c