schreibfaul1/ESP32-MiniWebRadio #647: h_sync Bild instabil

skassel2024, 2025-03-04:
Hi Wolle,
mir ist aufgefallen, dass bei Streams 64K aac teilweise das Bild horizontal "zittert" Bei anderen Streams mit 128K mp3 oder höher taucht das Problem nicht auf.
vG Sören

skassel2024, 2025-03-05:
Hi Wolle,
hab noch weiter getestet. Das Problem tritt anscheinend nur bei AAC Streams (64K) auf.
vG Sören

schreibfaul1, 2025-03-05:
Hi Sören,
das ist mir bisher nicht aufgefallen. Ich benutzte nur das Sunton Display, das ist das, was ich "umgebaut" habe.
Es scheint einen Zusammenhang mit dem PCLK Takt zu geben. Großer 11 MHz scheint die Lastgrenze zu sein. Getestet habe ich mit https://stream.studio21.ru/studio2196.aacp (96KBit). Das Elecrow Display kann ich auf 11MHz stellen und das Flackern ist verschwunden.
Beim Waveshare Display ist < 12MHz das Bild weg. Hab noch keine keine Ahnung warum.

schreibfaul1, 2025-03-06:
Ich habe es etwas optimiert.  Mit https://stream.studio21.ru/studio2196.aacp habe ich auf dem Waveshare Display seit Minuten ein stabiles Bild ohne Flackern.
1) Im AAC Dekoder sind alle Makros jetzt Inline-Funktionen.
2) menuconfig: PSRAM ohne ECC

Und getestet habe ich mit diesen Einstellungen: [docs/rgb displays/Waveshare 7' Display/Waveshare_7inch.txt](https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/V4-pre-release/docs/rgb%20displays/Waveshare%207'%20Display/Waveshare_7inch.txt)

skassel2024, 2025-03-07:
Hi Wolle,
mit den Änderungen ist zwar das Flackern weg, aber ein Umschalten der Sender dauert teilweise über eine Minute. mit
folgenden Fehlermeldungen:

[image]

Nach folgender Änderung in der common.h funktioniert die Senderumschaltung wieder, Fehlermeldungen sind weg, aber das Flackern ist auch wieder da:

[image]

Das Datenblatt des ST7262 System-On-Chip Driver for 800RGBx480
TFT LCD gibt folgende sync Werte an:

[image]

Das Flackern ist weiterhin nur bei AAC. Bei mp3 ist alles in Ordnung.

In diesem Zusammenhang ist mir aber noch folgendes aufgefallen (erst durch die Fehlermeldungen):
Bei DLNA werden keine Musikfiles gefunden. Da bin ich noch dran. Möglicherweise liegt das aber an den Freigaben. Oder hast Du da einen Tip?
vG Sören
[image]

schreibfaul1, 2025-03-07:
Hallo @Jason2866,
danke für deinen Beitrag. DMA für I2S und fürs Display bringen den S3 an die Grenze. Scheinbar lässt sich der Takt fürs Waveshare Display nicht unter 12MHz verringen. Bei meinem Sunton und Elecrow Display ist das kein Problem. Die Bildwiederholrate ist bei diesem Projekt nicht so entscheidend zumal in den Framebuffer geschrieben wird.

Das ist mein Waveshare mit 12MHz, leider kein Ton, ich habe kein DAC angeschlossen.
https://github.com/user-attachments/assets/09f2e9cf-0166-4be5-9b98-32949d83d93c

Jason2866, 2025-03-07:
Sieht so aus als dass du den Balance Akt hinbekommen hast! Meiner Erfahrung nach funktioniert 12Mhz mit allen RGB Boards. Ist ja auch noch in der Spezifikation.

skassel2024, 2025-03-07:
Hallo zusammen,
das Flackern hält sich nun in Grenzen. Benutze eben nur Streams mit mp3.
Mir gefällt das Waveshare wegen des Rahmens und der einfachen Einbauweise.
Wolle, wie könnte man die CPU-Auslastung über den Testmodus abfragen zusätzlich zu der Speicherauslastung?
vG Sören

[image]
[image]

https://github.com/user-attachments/assets/d9c9d108-4b5c-462b-ab32-8fc504c220d6

schreibfaul1, 2025-03-07:
Du kannst im Terminal rts<ENTER> eintippen und erhälst die "Run Time Statistics", also die Auslastung der Tasks, das sieht dann so aus:

[image]

Ich habe fas Display jetzt etwa eine Stunde am laufen mit dem aac "Studio21" Stream ohne Flackern. Bei AAC wird die CPU besondes gefordert. Während andere Codecs in einer Tabelle nachschauen oder nach "Kochbuch" verfahren ist es bei AAC anders. Der AAC Dekoder rechnet mit komplexen Zahlen.

Unser Waveshare Display hat zusätzliche ICs mit für CAN-Bus, RX484 und Portexpander. Ich kann mir vorstellen, dass es Beeinflussungen gibt, z.B kann der Protexpander das Display oder das TP zurücksetzen wenn es schlecht läuft.

skassel2024, 2025-03-07:
Welche Werte hast Du für hsync und vsync?
Bei 12Mhz ist zwar das Flackern weg, aber der Senderwechsel ist wieder extrem langsam. Daher bin ich wieder auf 13Mhz zurück.

skassel2024, 2025-03-11:
Hallo Wolle,
unter folgendem Link habe ich etwas interressantes gefunden: https://github.com/arendst/Tasmota/discussions/22553
Mit folgenden Werten habe ich ein flackerfreies Bild bei "Studio 21" und eine schnelle Senderumschaltung ohne Abstürze:

[image]
vG Sören

schreibfaul1, 2025-03-11:
Hallo Sören,
das hört sich gut an!
Ich habe ein flackerfreies Bild mit deinen Einstellungen, außer PCLK. Mein Display streikt bei 11,8MHz,
die unterste Grenze ist 12MHz und die Version ist  V1.1.

beste Grüße
Wolle

skassel2024, 2025-03-11:
Hi Wolle,
mein Display ist auch V1.1 mit ESP32-S3 N8R8. Bei Dir ist in platformio aber N4R8 voreingestellt.

meine Einstellung:

[image]

Habe versucht den .pixel_clock_hz noch weiter zu verringern. Bei 11,4Mhz ist auch bei mir schluss. 11,5Mhz geht noch.

skassel2024, 2025-03-14:
Hallo Wolle, 
seit 3 Tagen ohne flackern bei 11,5Mhz :-)))

Aber bei DLNA habe ich noch Probleme. Entweder timeout, oder es werden nicht alle Dateien im Ordner des Medienservers angezeigt. Bei mehrmaligem Aufruf (btn_DL_filelist) weden dann mal alle oder nur teilweise angezeigt. 

[image]
Beim zurückblättern kommt dann meist:
[image]
vG Sören

schreibfaul1, 2025-03-14:
Ja, habs auch,
das ist mei DLNA Server der Fritzbox:

[image]

ich habe noch keine Ahnung warum es nicht geht.

closed 2025-04-28 by github-actions[bot]