schreibfaul1/ESP32-MiniWebRadio #649: What display do you use?

linoreale, 2025-03-12:
Hello,
I'm using an ILI9488 display for this project but it seems not to be the best choice because touch pad panel only works in portrait mode, I tried all availlable options in the code.
So I decided to buy another display that works for sure.
In your experience what one are you using? Could you please tell me which 3,5" display works?
Thanks and regards.

valera39, 2025-03-12:
Hello,
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/638#issuecomment-2692290794 This kind of display is very capricious, it is better not to take it
https://aliexpress.ru/item/1005007461863045.html?sku_id=12000043381791367&spm=a2g2w.productlist.search_results.2.1fbf1848X6sCtp This is a link to the Russian segment of aliexpress. I only get these

Kicadder, 2025-03-12:
Hi all,
I use this one :
https://www.aliexpress.com/item/33057812377.html?spm=a2g0o.order_list.order_list_main.12.3a1c1802Xc4dsm
 
Correct seller and components AFAK.
Greetings!

closed 2025-04-26 by github-actions[bot]