schreibfaul1/ESP32-MiniWebRadio #650: Connection diagram

valera39, 2025-03-16:
Hello Dear Wolle!
I got the Sunton display. I don't understand the connection. I have marked with a question mark in the image. What is 16?
I'm looking forward to your clue!

schreibfaul1, 2025-03-16:
Great, these are the 16 colour lines, they are already connected on the display, HSYNV, VSYNC etc. too.
If Git is installed on your computer, simply go to "V4-pre-release"

[image]

there you will find the file for your display. Check whether the value 7 is set for TFT-CONTROLLER in common.h.
Then check whether under `TFT_CONTROLLER == 7 // RGB display` the correct values are set, change if necessary

Then compile as usual. The Sunton has a MONO DAC. You should then have a picture + sound.

You have to do the rest yourself.

schreibfaul1, 2025-03-16:
Important is the SD card in the branch there is a new "Content_on_SD_Card.zip" with larger graphics. It is best to use it to write to a new SD card.
The external DAC runs in parallel with the internal one.
You don't need to change anything on the display, the LCD, TP and audio work just as well.

valera39, 2025-03-16:
On the github page of your project I copied the link https://github.com/schreibfaul1/ESP32-MiniWebRadio.git. And in VSCode I can't select "V4-pre-release".

valera39, 2025-03-16:
Yes, I have Git installed, but I always use copying the link from the browser. Isn't that the right thing to do?
[image]

schreibfaul1, 2025-03-16:
If this does not work, simply select the branch, download the zip file and unzip it on your PC

valera39, 2025-03-16:
All right!
And a couple other things, I have an S3 N16R8 on display. In platformio.ini I select 
`board_build.partitions = boards/miniwebradio16MB.csv   ; 16MB Flash`and 
`board = ESP32-S3-DevKitC-1-N16R8   ; 16 MB quad Flash, 8MB octo PSRAM`, right?
I have set the AMP_ENABLE parameter to GPIO 10
I cut the track to the memory card connector and soldered to the 10 gpio, right?
[image]

valera39, 2025-03-16:
I did it!
[image]
It turns out that the track from gpio10 needs to be cut near the chip, not where I did it first.
The only thing that confuses me is the slow appearance and updating of menu icons on the display. The memory card speed is set to max and the card itself is class 10.
Many thanks for your help and developing a great radio, keep up the good work!
Looking forward to the menu from topic #591
 and other improvements.
Regards!

closed 2025-04-30 by github-actions[bot]