schreibfaul1/ESP32-MiniWebRadio #652: Can you check the stream, please?

valera39, 2025-03-17:
Hello Dear Wolle and all members of the community!
Please check the playback of this stream https://retroclub.hostingradio.ru/retroclub128.aacp. 
Playback is normal on the computer.
Thank you!

valera39, 2025-03-17:
I should add that on radio version 2.0 with VS1053 the stream works fine. Perhaps the broadcasters have changed something in the stream, so the author's help is required

skassel2024, 2025-03-17:
Hi valera39,
try: https://retroclub.hostingradio.ru/retroclub224.aacp

schreibfaul1, 2025-03-17:
Strange, the content type returned is `audio/mpeg`, which starts the MP3 decoder.
Correct is `audio/aac` 
The probable cause was the user agent.
The stream works if you install the latest version of the audioI2S library.

valera39, 2025-03-18:
It works with the actual libraries, thanks for the quick fix!

closed 2025-03-18 by valera39