schreibfaul1/ESP32-MiniWebRadio #654: Version 3.7-rc1.l   - Mar 17/2025

skassel2024, 2025-03-18:
Hi Wolle,
mit der aktuellen Version geht unter 13Mhz pixel_clock nix mehr: kein Senderwechsel, keine Reaktion auf Webinterface, kein DLNA.
Bei 13Mhz geht es wieder, aber das Flackern ist bei "Studio21" wieder da.

[image]

Beim Hochladen bleibt im Terminal connecting.... fast eine Minute, dann beginnt der Schreibprozess und ist nach wenigen Sekunden beendet. Radio funktioniert. DLNA ebenso.
vG Sören

[image]

schreibfaul1, 2025-03-18:
Ich habe mein Waveshare Display angeschlossen und alles sieht gut aus.
Den letzten "Commit" habe ich mit diesem Display angelegt, du kannst das 1:1 übernehmen. mich würde interessieren was bei dir passiert.

Alternative Idee:
Wenn du "Git Lens" als Erweiterung in VSCode installierst kann du zwischen den Versionen wechseln.

[image]
Das hellblaue ist der Branch, das blaue der Master.
Du kannst dann zurückgehen mit "Rebase current Branch onto Commit..." bis zu den Zeitpunkt wo es noch funktioniert hat. Das macht etwas Arbeit, weil du die common.h anpassen musst und du musst die "Änderungen verwerfen" wenn du zum nächsten Commit springen möchtest.

Das Terminal-Problem habe ich auch, mal sehen ob das beim nächsten Update von VSCode oder PlatformIO? wieder weg ist.

skassel2024, 2025-03-18:
Hi Wolle,
mit12Mhz geht nix
[image]

bei 13Mhz ok, ausser DLNA
[image]

in platformio hast Du 4MB Flash und 4MB PSRAM ausgewählt. Das Waveshare hat aber einen ESP32-S3 N8R8
eine Änderung 
[image]
[image]

werde noch Erase Flash und Full Clean probieren un gebe Dir Rückmeldung.
vG Sören

skassel2024, 2025-03-18:
nach Erase Flash und Full Clean keine Änderung, bei 12Mhz geht nix, bei 13Mhz ok, aber DLNA geht nicht. Audioplayer ok.
GitLens habe ich zwar installiert, kenne mich damit aber überhaupt nicht aus

schreibfaul1, 2025-03-18:
Das habe ich mir gedacht, FullClean und EraseFlash wird nichts ändern, und ob die Partitionstabelle 4 oder 8MB groß ist spielt nur für OTA eine Rolle.
Mein Display funktioniert immer noch tadellos mit 12MHz, hab somit keine Möglichkeit der Reproduktion.
Eine Vermutung ist, dass die zusätzliche HW des Displays irgend einen Einfluss hat. Schließlich ist jeder GPIO des Displays verwendet worden (vielleicht ein Reset des LCD oder TP oder ein IRQ?) Ich meine dass die zusätzlichen ICs auf dem Board wirklich "tot" sind.
Mit Gitlens kannst du in den Versionen zurückspringen. Du kannst so genau das Commit ermitteln an dem der Fehler auftrat.

skassel2024, 2025-03-19:
Hi Wolle,
habe jetzt das ganze an einem "jungfräulichen" Display mit 12Mhz am laufen. Allerdings noch ohne DAC. Geht soweit ich das überblicke alles bis auf DLNA. Ich glaube Du hast recht mit der HW. IO15 habe ich zwar an U6 Pin 1 abgelötet, aber da stört vermutlich noch was anderes was zu I2C gehört. Werde das in den nächsten Tagen testen und Dir berichten.
Aber was hast Du an Deinem Display alles verändert?
vG Sören

skassel2024, 2025-03-20:
Hi Wolle,
so ähnlich habe ich das auch gemacht, allerdings habe ich nur RX485 TX (IO15) unterbrochen. Da scheint das Problem zu liegen. Ich werde das noch für die anderen Pins nachholen. Vor Allem direkt am ESP32 und nicht irgend wo auf der Platine. Mir ist nur noch nicht klar, warum es mit einem "jungfräulichen" Display ohne Änderungen aber ohne den Anschluss des DAC funktioniert.
Bei DLNA vermute ich ein Timeout Problem. Verbindung zum media server wird unterbrochen? 
Wenn ich über die Weboberfläche DLNA aufrufe wird die Verbindung zum MiniWebRadio abgebrochen. Auch nach Neuladen der Seite kann ich nicht zum Radiomodus umschalten. Erst nach einem Reset.
Mit GitLens komme ich nicht klar. Ich kann zwar zwischen den Versionen umschalten, beim Hochladen kommen aber zu viele Fehlermeldungen über falsche Dateiversionen von Elementen. Mich da einzudenken ist mir noch zu kompliziert. Meine altbewährte Methode zip downloaden, entpacken und im Arbeitsbereich von VS Code einfügen hat bisher gut funktioniert.
Aber Danke für die Hinweise.
vG Sören

skassel2024, 2025-03-20:
Hi Wolle,
bin ein ganzes Stück weitergekommen. Bei Stromversorgung über UART1 Anschluss an PC funktioniert es bei 13Mhz, Bei Stromversorgung über USB an PC geht tonmässig nix. Bei Stromversorgung über Powebank (an UART1) kann ich bis 11,8Mhz runter. Dabei ist mir aufgefallen, dass das abhängig davon ist, ob ich das Gehäuse des ESP zusätzlich mit Masse verbinde. Meine Vermutung fehlende oder schlechte Verbindung zum Massepad unter dem ESP zur Platine. Mit der Masseverbindung bekomme ich auch Verbindung über DLNA.
Werde den Bereich Stromversorgung mir mal näher anschauen und weiter berichten.
vG Sören

Jason2866, 2025-03-20:
Es könnte auch sein dass die geringe Änderung der Versorgung Spannung einen nicht unerheblichen Einfluß darauf hat wie weit der Clock "runter gedreht" werden kann.
Wäre einen Versuch wert das mit einem gutem Labornetzteil mal zu prüfen.

closed 2025-05-04 by github-actions[bot]