schreibfaul1/ESP32-MiniWebRadio #656: Terminal connect . . .

skassel2024, 2025-03-23:
Hi Wolle,
nur zur Info; das mi der Terminalanzeige liegt möglicherweise an der IDF Version. Zurück zu ...rc1.zip ist die alte Terminalanzeige wieder da mit "Writing at ..."
[image]
Aber bei der neuesten IDF Version ...rc2.zip bleibt "connecting . . " stehen, bis der Screibprozess beendet ist. Das stört zwar nicht, es hat mich nur irritiert, da der Schreibfortschritt nicht mehr angezeigt wird.
vG Sören.

schreibfaul1, 2025-03-23:
Ja, das ist rproduzierbar. Während rc1 problemlos läuft kommt bei rc2 nach
`Connecting....`
gleich 
`Writing at 000030daf8... (100 %)`

der Ladefortschritt wir nicht mehr gezeigt.
Vielleicht ist das bei der nächsten Version wieder okay. Mal abwarten :-)

schreibfaul1, 2025-04-07:
Es gibt ein neues Esptool `esptoolpy v4.8.9` das diese Fehler beseitigt.
Danke an [Jason2866](https://github.com/Jason2866)
https://github.com/pioarduino/platform-espressif32/issues/132

skassel2024, 2025-04-07:
Super funktioniert wieder.

skassel2024, 2025-04-08:
ich habe das allerding ohne Deine Änderung in platformio.ini gemacht. Lediglich den Ordner packages in .platformio gelöscht, die aktuelle Version esptoolpy hat sich automatisch installiert. Umständlich, aber hat funktioniert.

[image]

[image]

Jason2866, 2025-04-13:
Ich habe ein silent pioarduino re-release der letzten Version gemacht. Die einzige Änderung ist das Update von esptool.py auf v4.8.9

closed 2025-05-28 by github-actions[bot]