schreibfaul1/ESP32-MiniWebRadio #657: Display ST7796 34 pins

linoreale, 2025-03-24:
Hi,
I bought this 3,5"display touch 34 pins module, ST7796, could you please tell me how to connect pins to this project?
Thanks and regards

[image]
[image]

schreibfaul1, 2025-03-24:
You need a display with an SPI input. The display in the photo probably has a parallel input.

If it is to be ST7796 then that would be applicable:

[image]

linoreale, 2025-03-25:
Hi, 
Is there a Converter that Can I use to connect this display to 5he radio?
Thanks

hengist60, 2025-03-25:
Hi linoreale

I am no expert in electronics (and I a sure I'll  be corrected if I'm
wrong😏)but a logic shift register such as

https://www.ti.com/product/CD74HC4094?utm_source=google&utm_medium=cpc&utm_campaign=asc-int-null-44700044290294005_prodfolderdynamic-cpc-pf-google-soas_en_int&utm_content=prodfolddynamic&ds_k=DYNAMIC+SEARCH+ADS&DCM=yes&gclsrc=aw.ds&gad_source=1&gclid=Cj0KCQjwqIm_BhDnARIsAKBYcmtBHrpl_x4dEg59xaU-H5xLudg6iq6v5tLKFiwAX5Fv90d73X2TVB4aAn2AEALw_wcB

(or two) might provide a way forward. It would be quite a bit of
flip-flopping around but if it were possible I am sure someone would have
already done it and a bit of googling around might be useful.

Regards

Jon Lloyd

closed 2025-05-09 by github-actions[bot]