schreibfaul1/ESP32-MiniWebRadio #659: V4-pre-release Compiling Error in main.cpp line 3132

TubaFonse, 2025-03-26:
Hallo Wolle, 
in der aktuellen SUNTON-Version vom 25.03. erhalte ich beim Compilieren folgenden Fehler: 

src/main.cpp:3132:32: error: no matching function for call to 'Audio::openai_speech(const char [11], const char [6], const char [8], const char [57], const char [4], const char [2])'
 3132 |             audio.openai_speech("openAI-key", "tts-1", "shimmer", "Today is a wonderful day to build something people love!", "mp3", "1");
      |             ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~      
In file included from src/common.h:52,
                 from src/main.cpp:1:
.pio/libdeps/esp32s3/ESP32-audioI2S/src/Audio.h:142:10: note: candidate: 'bool Audio::openai_speech(const String&, const String&, const String&, const String&, const String&, const String&, const String&)'
  142 |     bool openai_speech(const String& api_key, const String& model, const String& input, const String& instructions, const String& voice, const String& response_format, const String& speed);
      |          ^~~~~~~~~~~~~
.pio/libdeps/esp32s3/ESP32-audioI2S/src/Audio.h:142:10: note:   **_candidate expects 7 arguments, 6 provided_** 

(Die Version vom 23.03. hat noch funktioniert). 

Viele Grüße, 
Alfons

skassel2024, 2025-03-27:
Hi Wolle,
den Fehler habe ich auch. hier fehlt anscheinend ein Parameter:
In der aktuellen Audio.h ist "instructions" dazugekommen:
Audio.h
[image]

main.cpp
[image]

vG Sören

schreibfaul1, 2025-03-27:
Hallo,
der Kompilierfehler ist ein dummer Seiteneffekt der durch eine Änderung bei OpenAI-Speech in der audioI2S Bibliothek verursacht wurde.

Das wird hier nicht verwendet, zumal ein OpenAI-Key mit einem gültigen Abonnement Voraussetzung ist. Hinzugekommen ist ein "VIBE" Argument welches ich jetzt hinzugefügt habe.
Hier gibt es einige Beispiele:  https://www.openai.fm/

beste Grüße aus Berlin
Wolle

TubaFonse, 2025-03-27:
Jetzt läufts wieder, danke!!!
Liebe Grüße aus Tutzing, 
Alfons

closed 2025-03-27 by TubaFonse