schreibfaul1/ESP32-MiniWebRadio #660: v 4.0 no sound on the right channel

valera39, 2025-03-27:
Hi Wolle!
There is no sound from the right channel output of the PCM 5102A. I checked the soldering, contacts, swapped PCM 5102 from another radio of the third version, nothing helps. Sunton display, correct configuration
` #define I2S_DOUT           17
  #define I2S_BCLK           0
  #define I2S_LRC            18
  #define I2S_MCLK           -1  // important!` 
I haven't realized for days what I'm doing wrong?

[image]

valera39, 2025-03-27:
connection from GPIO 18 to LCK PCM 5102 is present, checked with an ohmmeter
Here are screenshots of the LCK, DIN and BCK oscilloscope

[image]

[image]

[image]
What else can be done to solve the problem?

schreibfaul1, 2025-03-27:
All 3 signals are perfect, so it can only be something on the PCM5102 board. Perhaps an interruption between the DAC and the jack socket?

valera39, 2025-03-27:
So the problem is not with the Sunton display, but with the DAC?
You and I have the same configuration and you don't have this problem, right?
I've tried taking the signal from both the connector and directly from the board pins, but to no avail.
I will now try again to take the DAC board from a working MVR v3.0 PCM 5102 and replace them with each other.

valera39, 2025-03-27:
In the balance and tone settings, the balance slider was pushed to the right as far as it would go, so there was no sound. I was obliged to pay attention to this, but I guess I've gotten stupid and old if I don't notice the obvious :)
Thanks for your input and advice and sorry for taking up your valuable time.
I hope I'm still allowed to ask questions in the future? :) 
Always looking forward to new achievements and new design of your radio.

schreibfaul1, 2025-03-28:
This is also a possible cause of the error. 😀

In the near future, it will be possible to set the WiFi password via the display. Then the entry via the AccessPoint and the "networks.csv" will no longer be necessary.

best regards
Wolle

closed 2025-03-28 by valera39