schreibfaul1/ESP32-MiniWebRadio #661: Standfuss Waveshare 7' Display

skassel2024, 2025-03-28:
hier mal eine Idee für einen Standfuss:

[image]

[DisplayHalter7inchZusammen.txt](https://github.com/user-attachments/files/19509464/DisplayHalter7inchZusammen.txt)
.txt durch .stl ersetzen

schreibfaul1, 2025-03-30:
Sehr saubere Arbeit! Ich kann erkennen, dass du eine weitere Leiterplatte eingebaut hast, wahrscheinlich DAC oder Vorverstärker oder beides.
Es sieht so aus, als wenn die Waveshare Displays gegenüber anderen Herstellern weiter verbreitet sind. Wenn du noch etwas zum Umbau des Displays beschreiben könntest würde ich das gerne archivieren und so anderen Usern zur Verfügung stellen.

skassel2024, 2025-04-04:
Hi Wolle,
ich versuche hier mal zu beschreiben, was ich an dem Waveshare Display geändert habe:
Die rot matkierten Leiterbahnen unterbrechen (mit einem feinen Skalpell).
Die grün markierten Pins sind die Anschlüsse für den DAC PCM5102MK 2.0
[image]

Als nächstes habe ich einen weiteren Steckverbinder an den I2C Steckverbinder angeklebt.
Im Bild unten die grün markierten Verbindungen hergestellt.
Dann noch die Brücke im grün markierten Bereich von 3V3 auf 5V umgelötet.
I2C VCC und I2C GND auf den zusätzlichen Steckverbinder gelegt (ebenfalls grün markiert).
[image]

So sieht dann der fertige Aufbau aus.
[image]

Und das ganze noch mit einem "Deckel" verschlossen.
Normalerweise hat man ein Gehäuse und baut dann das ganze dort ein. Ich bin einen anderen Weg gegangen und habe das Gehäuse um die Elektronik herum gebaut (3D Druck) um das schlanke Design des Displays zu erhalten.
[image]

Im Standfuß hätte noch ein Akku platz, aber das mache ich später vielleicht noch. Ein Test mit einer 5000mAh Powerbank (bei LIDL für knapp 7 EUR) ergab eine Laufzeit von ca. 4h. 
Hier noch die 3D Druckdateien. (.txt umbenennen in .stl
[DisplayHalter7inchZusammen.txt](https://github.com/user-attachments/files/19509464/DisplayHalter7inchZusammen.txt)
[DisplayHalter7inchDeckel.txt](https://github.com/user-attachments/files/19601649/DisplayHalter7inchDeckel.txt)

Was noch anzumerken ist: Bei meinem Board ist das Flachband zum Display abgeschirmt, was anscheinend nich bei allen Versionen so ist, wobei der Schirm eine undefinierbare Verbindung zu GND hat. Auf der Rückseiter der Platine berührt die Abschirmung teilwese das Blech auf der Rückseite des Displays, was bei mir zu den seltsamsten Fehlermeldungen geführt hat. Eine Isolierung der Abschirmung gegen das Blech, oder auch eine Dauerhafte Masseverbindung des Blechs zu GND der Platine hat hier Abhilfe gebracht und das Radio läuft stabil mit Deinen Einstellungen. Nun hoffe ich, dass ich bei der Beschreibung nichts vergessen oder übersehen habe.
Gerne darfst Du die Bilder, den Text oder die 3D Druckdateien weiter verwenden.
VG Sören

closed 2025-05-19 by github-actions[bot]