schreibfaul1/ESP32-MiniWebRadio #664: Clicks from DAC output in version 4

valera39, 2025-03-31:
Hi Wolle!
You and I have the same set of components, but I have from the output of PCM 5102 along with the useful signal periodically on any station there are clicks, emissions of DC voltage, which is clearly visible on the oscilloscope connected to the DAC output. If you press the "mute" icon on the display, the clicks stop, from which I conclude that the problem is in the DAC output.
What can it be and do you have such a problem?
Translated with DeepL.com (free version)

valera39, 2025-03-31:
I didn't manage to insert the video here, so I will leave a link to the file hosting
https://ru.files.fm/u/3r9v9zy7rw

valera39, 2025-04-05:
Dear Wolle, I am waiting for your answer or advice, my problem is not solved

schreibfaul1, 2025-04-06:
Hi valera,
I have absolutely no explanation for this. The sound is interrupted periodically. Perhaps a wire (BCLK, DOUT, LRCK) is touching a neighbouring pin or there is a problem with the power supply. I think the wiring looked good in your photo.  And we have the same display.

[image]

 The DAC on the display will continue to work. What about the "speak" connector?

valera39, 2025-04-07:
Hi Dear Wolle!
I apologize for not replying right away, had a lot of urgent things piling up, just checked out your tips today.
When powered by battery (power bank), both with external and internal DAC playback is perfect But when powered from an external source, there is a clicking noise. This is especially pronounced with my configuration: PCM5102 and the display is powered from 5 volts and the power amplifier from 12 volts. I think you are right, it's the power supply. What do you recommend?

schreibfaul1, 2025-04-07:
You have found the cause of the crackling, which is good news. I am sure that you can now solve the problem and I wish you every success.

closed 2025-05-23 by github-actions[bot]