schreibfaul1/ESP32-MiniWebRadio #665: DLNA The Server is not responding

skassel2024, 2025-04-04:
Hi Wolle,
folgende Meldung "... not responding" kommt sporadisch. Nach mehrmaligem Versuch wird der Inhalt des Media Servers aufgelistet. Gibt es da eine möglichkeit "Time Out" zu verlängern, möglicherweise braucht der MediaServer zu lange für eine Antwort. Oder was mache ich falsch?
vG Sören
[image]

schreibfaul1, 2025-04-05:
Hi Sören,
im DLNA Server habe ich die Zeit bis zum Timeout von 4000ms auf 6000ms erhöht. Allerdings glaube ich nicht dass das die Ursache ist. Ganz allgemein liegen die Antwortzeiten im Heimnetz bei etwa 2ms.
Ich habe die TCP ReceiveMailbox verringert, das könnte sich positiv auswirken, kleinere Mailbox -> kleinere Antwortzeit.
Eventuell kann es an der SD Karte liegen, obwohl es nicht offensichtlich ist. Ich habe eine SD Karte mit 16GB von Gigastone die genau solche Fehler verursacht. Wenn du die Möglichkeit hast, wechsle die SD Karte gegen eine von einem anderen Hersteller testweise aus.

Ansonsten habe ich in etwa die gleiche Konfiguration wie du, den DLNA Server der FritzBox und einen selbst aufgesetzten auf meinem PC, allerdings ohne "... is not responding"

skassel2024, 2025-04-05:
Hi Wolle,
danke für den Tip mit der SD-Karte. Ein Austausch hat in Bezug DLNA zwar nichts gebracht. Immer noch "..is not responding". Allerdings ist der Start des Radios wesentlich schneller geworden. Vorher dauerte es zwischen "SD card found" und "seek for WiFi" bis zu 10sec.

DLNA geht leider noch nicht, dafür weitere seltsame Meldungen:

[image]

vG Sören

skassel2024, 2025-04-06:
Hi Wolle,
mit der aktuellen Version (6.04.2025) funktioniert DLNA bei beiden, Fritz! und PC. Ein Neustart und der Bildaufbau sind richtig schnell geworden.
vG Sören

[image]

[image]

[image]

closed 2025-05-21 by github-actions[bot]