schreibfaul1/ESP32-MiniWebRadio #670: Wifi problem

Gyula1984, 2025-04-14:
Hi i installed the V4 version when it loaded it shows the available WiFis but i cannot connect to them, What am I doing wrong? Thank you

skassel2024, 2025-04-15:
Entering the password via the display doesn't work for me at the moment either. So I enter my SSID and password in common.h. Network.csv is no longer used.
[image]

Gyula1984, 2025-04-15:
thanks i have tired but pop up this menu when i started the radio, maybe the board is broken, this is my second board

schreibfaul1, 2025-04-15:
I have changed something else and hope that it works better now.

[image]

The number of green circles is the number of characters in the password. Is there a difference?
There may be remnants of old installations in the NVS. Please delete everything with `Erase Flash`. The terminal must be closed beforehand, otherwise it will not work.

Gyula1984, 2025-04-15:
i will check it later thanks the answer

skassel2024, 2025-04-15:
Hi Wolle, nach Erase Flash und Eingabe des PW über das Display sieht das bei mir so aus. OK?
[image]

schreibfaul1, 2025-04-15:
Nö, so soll es nicht aussehen. Das ist verwirrend. Die KI meint `char pass[64 * 4] = {0}; ` wäre nicht sicher.
In deinem Bildschirmfoto gibt es noch irgendwas anderes.  🟢 ist wahrscheinlich nicht NULL-terminiert wenn strcat() verwendet wird?
Habs noch mal verändert.
vG Wolle

Gyula1984, 2025-04-15:
Ok my board has gone but only the wifi connection and this is my second board which died, maybe china quality i dont know, thanks all, little bit strange it can found wifi networks but can not connect

schreibfaul1, 2025-04-15:
I don't think your board is faulty. It could be a problem with the antenna.

[image]

It doesn't have to look like the picture, but a solder bridge may still be necessary.
You can increase the transmission power in common.h.

[image]

Good luck!

Gyula1984, 2025-04-15:
thanks a lot it works!

steve6375, 2025-04-15:
What worked, changing the antenna link or increasing the power setting? If power, what setting did you use?

Gyula1984, 2025-04-16:
i have changed to 21 the power setting

skassel2024, 2025-04-16:
Hi Wolle,
nach Deinen Änderungen und erneutem Erase Flash sieht es immer noch so aus. Radio verbindet sich mit Router und funktioniert.
[image]

 das PW in Klartext wird richtig angezeigt  (Zeile 1285 in main.cpp eingefügt) 
[image]

vG Sören

valera39, 2025-04-16:
Hi Wolle,!
How do I properly connect to the right network? The “OK” button is inactive and pressing it after entering the password does not result in anything. After entering the password, I pressed the “RET” button, the radio rebooted and only after the reboot did the connection happen. Then why the “OK” button?

valera39, 2025-04-16:
I haven't connected an external DAC and I2C bus yet

skassel2024, 2025-04-16:
Hi Wolle,
hier mein hexdump:
[image]

schreibfaul1, 2025-04-16:
Hi Sören,
sehr gut, bester Hexeditor den es kostenlos gibt! Damit kann man gut viele Dateiformate analysieren.

Also ist zwischen den grünen Kreisen befindet sich 0x20, das ist das Leerzeichen.

[image]

Jetzt ist die Frage, was das da zu suchen hat. Muss noch überlegen....
Aber erst mal vielen Dank!

skassel2024, 2025-04-17:
Hi Wolle,
ich denke das hängt mit der Unicode Version zusammen. Dein greenDot f09f9fa2 ist Version12.
Mit  dem Code 🔵 LARGE BLUE CIRCLE (Version6) funktioniert es. Aber ich habe keinen grünen Punkt in der Vers. 6 gefunden, daher der blaue Punkt. 
vG Sören

[image]

schreibfaul1, 2025-04-18:
Grün macht Probleme und blau funktioniert?
Ich habe in Windows11 VSCode installiert um zu testen, ob es eine Abhängigkeit zum Betriebssystem gibt. Das konnte ich nicht feststellen.

Jetzt gibt es blaue Kreise für das Passwort. Das was ich machen möchte ist die Emojis (nur Kreise und Quadrate) in den Text zu integrieren um sie auf das Display zu zeichnen. An Stelle eines Buchstabens wird dann z.B. ein gelbes Quadat gezeichnet. Aber soweit bin ich noch nicht.

viele Grüße 
Wolle

skassel2024, 2025-04-18:
Hi Wolle,
ich benutze noch Win10. Ich vermute doch, dass das damit zusammen hängt. Ich konnte auch nicht herausfinden, ob das mit den Terminaleinstellungen in platformio (V3.3.4)  etwas zu tun hat.
Jedenfalls nicht mit Deinem Code. Der grüne Punkt hat unicode Version 12 und der blaue unicode Version 6. Da auch das PW aus dem Flash richtig angezeigt wird, kann das doch nur mit der Terminaldarstellung von platformio zusammenhängen?
vG Sören

skassel2024, 2025-04-18:
Hi Wolle,
nur die beiden rot und blau (unicode V6) funktionieren. Bei allen anderen (unicode V12) sind wieder die Leerzeichen dazwischen.

[image]

schreibfaul1, 2025-04-18:
Das wird die Erklärung sein. Es ist nicht ganz so wichtig wie das Log aussieht. Und wenn es mit blauen Kreisen funktioniert ist es gut so.
Ich möchte dass die "Emojis" auf den Display gezeigt werden.
Ein erster Test:

[image]

[image]

skassel2024, 2025-04-18:
Ja, das ist gut so und nicht wichtig, nur ein "Schönheitsfehler". Aber mich würde interressieren, warum es bei meiner Konfiguration (Betriebssystem, Terminal in Platformio) nicht funktioniert, Wenn ich da was gefunden habe, gebe ich Rückmeldung. 
vG Sören

skassel2024, 2025-04-21:
Hi Wolle
mit einem externen Terminalprogramm (PuTTY) ist die Darstellung ohne die Leerzeichen. Folglich liegt es am integrierten Terminal oder dessen Einstellungen von Platformio.
Leider habe ich noch keine Einstellmöglichkeit gefunden um Unicode12 korrekt darzustellen. Zu Auswahl stehen nur Unicode 6 oder 11. VS Code und Platformio sind auf dem aktuellen Stand. Ich versuche das noch auf einem Rechner mit win11.
vG Sören
[image]

schreibfaul1, 2025-04-21:
Kann sein, dass es am integriertem Terminal liegt. Die Leerfelder zwischen den Kreisen sehen eher zufällig aus, ich kann keine Regelmäßigkeit erkennen. Somit könnte es ein zeitliches Problem sein. Was passiert wenn die Geschwindigkeit erhöht oder reduziert wird?

skassel2024, 2025-04-21:
Ein reduzieren der Baudrate des Terminals hat nix gebracht. Aber warum funktioniert es bei Dir? Doch Win10 die Ursache?
Steh momentan total auf dem Schlauch:-((
Ausgabe auch bei 9600baud:

[image]

closed 2025-06-05 by github-actions[bot]