schreibfaul1/ESP32-MiniWebRadio #672: feature request(add tag to stations list)

clear-sky, 2025-04-15:
hi dear Wolle!
thanks for your efforts for this great project!
this is only a suggestion:
In my opinion, for better station management, it would be better to add a tag to the station list so that each station can be assigned a tag.
Tags add information about the station's genre, such as news, pop, jazz, etc.
There are usually four colored buttons on most remote controls: green, blue, red, and yellow.
may be adding this keys be useful to select major tags

clear-sky, 2025-04-15:
another idea: A key can be introduced as a tag on the remote that can be pressed to select a specific tag.

schreibfaul1, 2025-04-15:
The Radio Browser provides tags, certainly over 100 different ones, and several can be assigned to one station.

[image]

I don't think that's easy to realise.

closed 2025-04-15 by clear-sky

schreibfaul1, 2025-04-24:
Alternatively, you could assign a symbol to the station name.
These are the symbols that are possible:  🟢🟡🔴🔵🟠🟣🟤🟩🟨🟥🟦🟧🟪🟫
However, the symbol is displayed everywhere, not just in the list.

[image]

[image]

[image]

clear-sky, 2025-04-24:
Hi Dear Wolle!
very very nice ! thank you
another idea: is that possible to have several station list as favorite1-2-etc?

clear-sky, 2025-04-24:
another idea: mixing with your tags
assume you add a red circle to some stations!
can user select red circle through another menu and access to station list that have red circle tag?