schreibfaul1/ESP32-MiniWebRadio #673: GIF nettes Gimmick

skassel2024, 2025-04-16:
Hi wolle,
ist das der Beginn zu ESP32-MiniWebTV ;-)
möchte einfal nur mal Danke sagen für Deine tolle Arbeit.
[image]

schreibfaul1, 2025-04-16:
Danke,
den GIF Dekoder gibt es schon lange, allerdings gingen nur Standbilder. Das habe ich jetzt geändert um ggf. animierte Icons einzubauen. Übrigens habe ich dieses GIF zum testen benutzt:

[image]

Das ist Briefmarkenfernsehen. Auf den großen Displays geht es besser, die kleinen sind wegen dem SPI zu langsam.

beste Grüße
Wolle

valera39, 2025-04-16:
Did I miss something? What gif image? I don't have anything like that in the latest version.

schreibfaul1, 2025-04-16:
Don't worry, you haven't missed anything. I have improved the GIF decoder to add animated icons in the future.
You can save my GIF on the SD card in common/l/ (with a large display, otherwise common/m or /s) and if you then type gif <enter> in the serial terminal it will be played.

best regards
Wolle