schreibfaul1/ESP32-MiniWebRadio #675: Question: Bluetooth (BT)

pisascha, 2025-04-20:
Hi Wolle, 
just two short questions. I want to expand my radio with the bt emitter. 

The pin layout is 1.54, do you use an adapter to 2.56?

Do I have to solder the cable to one of the two sides or doesn't matter?

[image]

Thank you for your help in advance!

schreibfaul1, 2025-04-20:
That's how I did it. You don't have to do it this way. Please make sure that there is no contact between the pins before you install the BT_Emitter. I have already broken a board this way.

[image]

pisascha, 2025-04-20:
Thank you. The cable is soldered to RX, is this required?

pisascha, 2025-04-21:
Short feedback: used an old Adapter and works fine. 

[image]

The bt device (speaker) is Not recognized after switching it off.  After repowering the speaker I have to start the pairing again. But only the speaker, not the emitter.

Thanks a lot for your help!

closed 2025-06-05 by github-actions[bot]