schreibfaul1/ESP32-MiniWebRadio #677: Not doing anything after programming

jedi11csa, 2025-04-24:
Hi 
I am using VS Code - Platform IO, all up to date.
I have an [ESP32-S3-DevKitC-1 16MB Flash + 8MB PSRAM.](https://lonelybinary.com/collections/esp32/products/s3?variant=43784065712285)
TFT Touch screen is [ILI9488.](https://www.ebay.com.au/itm/315674301885?var=614092694939)
And I2S Stereo DAC Decoder is the [PCM5102.](https://lonelybinary.com/collections/i2s/products/i2s-stereo-dac-decoder-pcm5102-module)

Cloned the project, following the instructions:
- Change the environment to "env:esp32s3 (ESP322-MiniWebRadio)"
- In "platformio.ini" under `[esp32s3]` I have `board_build.partitions = boards/miniwebradio16MB.csv   ;16MB Flash` uncommented. And, `board = ESP32-S3-DevKitC-1-N16R8   ;16 MB quad Flash, 8MB octal PSRAM` uncommented.
- In 'common.h' changed values of "_SSID" and "_PW" to match my WiFi credentials.
- Also in 'common.h' changed values of the fields; TFT Controller, TFT_Rotation, TP_ROTATION, and TP_Version, to match my set-up. `#define TFT_CONTROLLER   4  //(4)ILI9488`, `#define TFT_ROTATION   3 // only SPI displays, 1 or 3 (landscape)`, `#define TP_ROTATION   3 //  only SPI displays, 1 or 3 (landscape)` `#define TP_VERSION  4 // only SPI displays (4)ILI9488`.
- I have extracted the "Content_on_SD_Card.zip" to an SD Card, and put it into the reader on the back of the TFT Screen board.
- Also have removed the series resistors replacing them with solder bridges.
- Lastly, have followed the wiring schematics, wiring the DAC, Screen+SD-Card, and ESP32-S3 together.

I build the project. Upload/program it to the ESP32-S3 on it's 'UART' USB-C port. And nothing... I see power LEDs on the ESP32-S3, and the PCM5102. And I measure +3V3 on the SD Card too. but the ESP32-S3 doesn't show up on my Routers WiFi list, and there's no life on the TFT screen.

In "platformio.ini" under `[esp32s3]`, I've tried changing the options `board = ESP32-S3-DevKitC-1-N16R8   ;16 MB quad Flash, 8MB octal PSRAM` to `board = ESP32-S3-DevKitC-1-N16R8V  ;16 MB octal Flash, 8MB octal PSRAM`, just to see if that would work, but it didn't work either.

Can you help me please?
I'd love to get this project working, as it's very intriguing and cool :)

JakesESP32, 2025-04-24:
Open a serial monitor and check if the SD card is being read the moment you do a reset, that's most probably your problem.

jedi11csa, 2025-04-24:
Hi JakesESP32, thanks for helping.

Here's a copy/paste of the results:

```ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x104c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xaec
load:0x403cb700,len:0x2cfc
entry 0x403c889c

ESP32 Chip: ESP32-S3
Arduino Version: 3.2.0
ESP-IDF Version: 5.4.1
MiniWebRadio Version 4.0-rc2   - Apr 18/2025
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 5555620 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V4     *
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
E (231) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (232) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[   238][E][SD_MMC.cpp:295] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
00:00:00 SD Card Mount Failed
```

So, looks like I need to solder in some pull-up resistors. The back of my ILI9488 TFT-Screen+SD-Card-Reader looks like the picture on the [main page](https://github.com/schreibfaul1/ESP32-MiniWebRadio) "Known Problems" section, the picture of the red coloured pcb with the resistors to remove.

My questions would be:
- Will 10K resistors do for pull-ups?
- Do I put one on the SD_MOSI trace, one on the SD_MISO trace, and one on the SD_SCK trace? Or just one on the SD_MOSI trace, and one on the SD_MISO trace?
- Appart from this issue, was there anything else you think I might need to try? Or was it just the missing pull-ups the Serial Monitor message states?

Again, thanks for chiming in!! :D

JakesESP32, 2025-04-24:
As expected ....

Frankly, I did not bother to get the display onboard SD reader working. All I did is I used an external SD card reader, I did not have to change any resistors on that.

Not very efficient, but it least it got to work. Nothing else was needed, the project works exactly as published apart from this speed bump.

Kind regards

jedi11csa, 2025-04-24:
I see.

Have just tried with the SD Card reader on the back of the TFT Screen. Three 10K resistors. One to each line, as stated in my questions.

No dice, unfortunately. Same error code.

Might have to try with an external SD Card reader like you did.

Thanks again for the help! Much appreciated :D

jedi11csa, 2025-04-27:
I bought an [SD-Card Reader](https://www.jaycar.com.au/arduino-compatible-sd-card-interface-module/p/XC4386) and hooked it up.

After much frustration, with the same origianl error:
```E (231) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (232) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[   238][E][SD_MMC.cpp:295] begin(): Failed to initialize the card (0x107). Make sure SD card lines have pull-up resistors in place.
00:00:00 SD Card Mount Failed
```
I tried desoldering the 10K pull-up resistor on this external board, connected to the MISO (D0) trace. But that would sometimes produce some kind of hit-n-miss result, or not. Resoldering it, I got a similar weird "hit and miss" results.

The 'hit and miss' result would see the project load in the serial monitor. It found my WiFi, and I could sort of load the webserver. But images and a lot of functionality wouldn't work. So, more a miss than hit. But this was a start.

Tried various SD cards. 
-  SanDisk Extreme 32GB
- SanDisk Ultra 32GB
- Lexar 300x 64GB

All produced the above 'hit or miss' result, or the original error, "Make sure SD card lines have pull-up resistors in place.", which was unfortunate. Especially as I'd now resoldered the 10K pull-up on this external SD-Card reader board.

When the 'hit or miss' happened, within the serial monitor readouts, most times, the `settings.json` file couldn't be found or loaded. As well as a lot of image files too.

So I tried changing the `#define SDMMC_FREQUENCY  80000000` value. First from 80MHz to 40MHz; Without success. In fact this just provoked the original error to happen, and no 'hit or miss' results at all.

But, what has seemed to work, somewhat, is changing the value to 20MHz:
`#define SDMMC_FREQUENCY         20000000                        // 80000000 or 40000000 Hz`

I could see the webserver page in Firefox. Change the country location. See my pre-loaded radio stations, and the music in the `audiofiles` folder on the SD card too.

I could actually hear the radio stations stream music!! And play a track from the SD card as well!! Woo-hoo!

**Further Issues:**

1. The webserver doesn't always display the radio station icons I've pre-loaded into the "l", "m" and "s" folders. They appear when using the arrows to change the selection, but when the music starts, they disappear, replaced by 'the smurf' icon.
2. Clicks heard on the audio. There's an issue topic before my issue topic addressing this. And my replication of that issue is probably because this whole thing is held together by "string, chewing-gum and sticky-tape" on a breadboard at the moment. Which brings me nicely to the next point...
3. Waiting on a fab house to manufacture and deliver a PCB I designed for this, so I can rid myself of the blasted breadboard. Hoping a pcb will do the trick for issue 2.
4. The TFT-Display doesn't work. It flickers a white screen. That's it. Hopefully getting my pcb from the fab house and proper electrical connections will address this issue. 

Been fun, thus far. Thanks to all involved for the frustration and the excitement, haha!! :D

JakesESP32, 2025-04-27:
Glad to see you made such good progress. 👍 

Regarding the TFT white screen, try tying the display RST pin to 3V3, I was able get it to work on the breadboard by doing that.  

At the frequencies present here, a breadboard and prototyping wires does produce sometimes strange results.

Keep on having fun !!! 👍

jedi11csa, 2025-05-02:
SUCCESS!!!! 🎆🎉

Got my PCB's and soldered in some female header strips. Placed the ESP32-S3. The PCM5102. The external SD Card reader (with all original resistors in place). And of course, the Touch screen ILI9488 TFT display.

Programmed the ESP32-S3, and used the SanDisk Extreme 32GB SD-card, with the SSMMC_Frequency value set at 20MHz.

Thing loaded up, and from my previous attempts via the webserver, it knew I was in sunny Australia, showed the flag, the right time (I changed the ntp servers with the platformio.ini file), **then started playing music from Double J!!!!**

Used the touchscreen to change channels. All radio stations I preloaded played through the speakers clearly, crisp, awesome!

Tried out the SanDisk Ultra 32GB SD-card. That had issues. So tried changing the SDMMC_Frequency to 40MHz. That fixed it.

**_Issues:_** 

- I changed the Wallpaper.jpg image. Neither SD-card/SSDMMC_Frequency-setting would display it on the TFT Display. Don't know why. Found the code in the main.cpp, and it calls for `drawImage("/common/Wallpaper.jpg", 0, 0); // Wallpaper` So tried moving the file into the common folder, as it's in the common/l, common/m, and common/s folders (all images sized appropriately for their respective folder). Even though the "MiniWebRadioV4.jpg" is similarly found in all "common/l", "common/m", and "common/s" folders. Dunno? My wallpaper don't wallpaper on this?? Love to fix it though ;)
- My radio station logos I copied into the "l"/"m"/"s" folders within the "logo" folder (again, all appropriately sized images) wouldn't show on the display. Bummer... again, would like to fix that up so they show on the display.
- One good point to counter the last, on the webserver, the logos show up, and stay put. So, "win!" :D
- The only red coloured serial-monitor warnings that show up are "AUDIO_info: file "/common/m/unknown.png" not found. But that's because the file is a jpg. Dunno, it did show up on the webserver page one time, briefly, before being replaced by the radio station logo.
- Tunes and tune-playlists I have in the "audiofiles" folder, don't seem to play at all.

Still, wanted to primarily hear radio on this. And I can; So, a win for the good guys!! Love it!! ❤🙏👍😁

JakesESP32, 2025-05-02:
👍😁

closed 2025-05-26 by jedi11csa