schreibfaul1/ESP32-MiniWebRadio #678: BH1750 in version 4

pisascha, 2025-04-25:
Hello Wolle,

are Pin 41 and 42 in version 4 still regulating the light sensor?

There are comments in the code that are related to the touchpad or is the light sensor bounded to the touchdisplay because of the brightness?

[image]

Sorry to bother you with such questions, I'm new to this

schreibfaul1, 2025-04-26:
Yes, if you have a light sensor you can connect it.
The sensor is recognised automatically.
Without BH1750, the display brightness can be adjusted using the slider.
With sensor: The BH1750 is dimmed and the display brightness at night is adjusted using the slider.

omegaxxl, 2025-06-03:
Hi,
my BH1750 with ST7796 display is not working. I couldn't find a menu where I could adjust the brightness. Am I doing something wrong or is the display not capable of that?

schreibfaul1, 2025-06-06:
I can't answer that question. There are displays that do not have a pin for the backlight.

omegaxxl, 2025-06-06:
Sorry, I probably wrote it wrong. I have a regular TFT display with an ST7796 controller, which only has a pin for the LED. It has the same pins as yours in the picture in the docs folder ILI9488_pins.jpg

schreibfaul1, 2025-06-06:
Okay, then the brightness control will certainly work with the slider.
The BH1750 communicates via the I2C bus (SDA/SCL)

This is the code in setup():
````c++
    if(I2C_SCL != -1){
        _f_BH1750_found = BH1750.begin(BH1750.ADDR_TO_GROUND , I2C_SDA, I2C_SCL);// init the sensor with address pin connetcted to ground
    }
    if (_f_BH1750_found) {                                                  // result (bool) wil be be "false" if no sensor found
        SerialPrintfln("setup: ....  " ANSI_ESC_WHITE "Ambient Light Sensor BH1750 found");
        BH1750.setResolutionMode(BH1750.ONE_TIME_H_RESOLUTION_MODE);
        BH1750.setSensitivity(BH1750.SENSITIVITY_ADJ_MAX);
    }
````
BH1750.ADDR_TO_GROUND is 0x23
BH1750.ADDR_TO_VCC is 0x5C

Perhaps the address is simply the wrong one, this can be set on the boards.

Otherwise you could log what happens on the I2C bus to narrow down the error.

schreibfaul1, 2025-06-07:
If the sensor is found, the I2C communication is certainly OK.
Maybe the brightness symbol is grey, then the TFT_BL is not assigned to a GPIO

[image]

omegaxxl, 2025-06-08:
Hi, I think I have a different version. My settings look like this. I haven't found a picture of the settings like here anywhere.

[image]

omegaxxl, 2025-06-08:
Hi, the problem is solved. There were two basic errors. I was looking for settings on the web interface. And the second error was in the display. I couldn't set the touch layer. I tried another display and everything started working.
Finally, I tried different settings on the first display and ended up with this strange setting for me with which the touch layer works correctly. 
Thank you very much for your help.

[image]

schreibfaul1, 2025-06-09:
Yes, the settings sometimes need to be adjusted. Good luck with your project.