schreibfaul1/ESP32-MiniWebRadio #679: task_wdt: Task watchdog got triggered - Endlos-Schleife

TubaFonse, 2025-04-26:
Hallo Wolle, 

nach ca. 30 Minuten Spielzeit erhalte ich folgenden Fehler (Endlosschleife - nur durch Hardware-Reset / Power-Off/-On zu beenden). 
Ich versuche mal, ob der Fehler auch mit einem anderen Sender auftritt: 

00:00:00 Country: ... AT
00:00:00 icy-descr:   Arabella Austropop
00:00:00 StationName: Arabella Austropop
00:00:00 icy-url: ..  https://www.arabella.at
00:00:00 AUDIO_info:  AACDecoder has been initialized, free Heap: 154835 bytes , free stack 5880 DWORDs
00:00:00 lastURL: ..  https://edge05.streams.arabella.at/arabella-austropop
17:04:12 Country: ... AT
17:04:12 Audiocodec:  AAC
17:04:12 ftpServer:   Ftp server waiting for connection on port 21
17:04:13 BT-Emitter:  KCX_BT_Emitter found 
17:04:13 AUDIO_info:  stream ready
17:04:13 AUDIO_info:  syncword found at pos 0
17:04:13 AUDIO_info:  Channels: 2
17:04:13 AUDIO_info:  SampleRate: 44100
17:04:13 AUDIO_info:  BitsPerSample: 16
17:04:13 AUDIO_info:  BitRate: 85788
17:04:13 AUDIO_info:  AAC HeaderFormat: ADTS
17:04:13 AUDIO_info:  Spectral band replication: upsampled SBR
17:04:14 BT-Emitter:  Status -> Scan...
17:04:14 BT-Emitter:  Status -> Disconnected
17:04:14 BT-Emitter:  POWER ON 
17:04:14 BT-Emitter:  POWER ON
17:04:15 BT-Emitter:  Version KCX_BT_RTX_V1.4
17:04:15 AUDIO_info:  slow stream, dropouts are possible
17:04:15 StreamTitle: GEORG DANZER & WILFRIED - SCHAU SCHATZI
17:04:15 BT-Emitter:  Volume -> 0
17:04:16 BT-Emitter:  wrong command: AT+VOL=0
17:04:17 BT-Emitter:  scanned: MacAdd:449f4910164a,Name:WZ-BT5.0
17:04:17 BT-Emitter:  CON LAST
17:04:20 BT-Emitter:  Status -> Connected
17:04:21 DLNA_server: [0] 192.168.178.1:49000  FRITZ7490
17:04:21 DLNA_server: [1] 192.168.178.24:50001  DS213J_AL
17:04:21 DLNA_server: [2] 192.168.178.29:2869  WIN10ALF: AH:
17:04:21 DLNA_server: 3 media server found
17:05:21 StreamTitle: CARL PEYER & GISI HAFNER - HERZKLOPFEN
17:08:22 StreamTitle: PIZZERA & JAUS - DIALEKT'S MI
17:11:11 StreamTitle: WIENER WAHNSINN - NOCH SOVü JOA
17:16:41 StreamTitle: WIR4 - TRäNEN TROCKNEN SCHNELL LIVE
17:21:30 StreamTitle: JOSH. - RING IN DER HAND
17:23:57 StreamTitle: ERNST MOLDEN & WILLI RESETARITS - RUDSCHDUAM
17:27:09 StreamTitle: WIR4 FEAT. WOLFGANG AMBROS - LAÃx MI AMOI NO D'SUNN AUFGEH'N SEGN LIVE
E (1408154) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (1408154) task_wdt:  - IDLE1 (CPU 1)
E (1408154) task_wdt: Tasks currently running:
E (1408154) task_wdt: CPU 0: IDLE0
E (1408154) task_wdt: CPU 1: loopTask
E (1408154) task_wdt: Print CPU 1 backtrace

Backtrace: 0x403775F2:0x3FCA21F0 0x4037BEF9:0x3FCA2210 0x4207AD9A:0x3FCD2840 0x42084BC2:0x3FCD2890 0x42006F55:0x3FCD2A40 0x420070C5:0x3FCD2A90 0x4202D105:0x3FCD2AB0 0x420999BC:0x3FCD2B70
  #0  0x403775F2 in esp_crosscore_isr at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf\components\esp_system/crosscore_int.c:73
  #1  0x3FCA21F0 in _xt_exception_table at ??:?
  #2  0x4037BEF9 in _xt_lowint1 at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf\components\xtensa/xtensa_vectors.S:1240
  #3  0x3FCA2210 in _xt_exception_table at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:67       
  #4  0x4207AD9A in TFT_SPI::analyzeText(char const*, unsigned short*, unsigned short*, unsigned short) at lib/tftLib/tft_spi.cpp:1476
  #5  0x42084BC2 in TFT_SPI::writeText(char const*, unsigned short, unsigned short, short, short, unsigned char, unsigned char, bool, bool, bool) at lib/tftLib/tft_spi.cpp:1795
  #6  0x42006F55 in textbox::writeText(char const*) at src/common.h:1851
  #7  0x420070C5 in showStreamTitle(char const*) at src/main.cpp:844
  #8  0x4202D105 in showStreamTitle(char const*) at src/main.cpp:835
      (inlined by) loop() at src/main.cpp:3072
  #9  0x420999BC in loopTask(void*) at C:/Users/Hoeckmayr/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:74

E (1413154) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (1413154) task_wdt:  - IDLE1 (CPU 1)
E (1413154) task_wdt: Tasks currently running:
E (1413154) task_wdt: CPU 0: PeriodicTask
E (1413154) task_wdt: CPU 1: loopTask
E (1413154) task_wdt: Print CPU 1 backtrace

Backtrace: 0x403775F2:0x3FCA21F0 0x4037BEF9:0x3FCA2210 0x4207ADA3:0x3FCD2840 0x42084BC2:0x3FCD2890 0x42006F55:0x3FCD2A40 0x420070C5:0x3FCD2A90 0x4202D105:0x3FCD2AB0 0x420999BC:0x3FCD2B70
  #0  0x403775F2 in esp_crosscore_isr at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf\components\esp_system/crosscore_int.c:73
  #1  0x3FCA21F0 in _xt_exception_table at ??:?
  #2  0x4037BEF9 in _xt_lowint1 at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf\components\xtensa/xtensa_vectors.S:1240
  #3  0x3FCA2210 in _xt_exception_table at C:\Users\Hoeckmayr\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:67       
  #4  0x4207ADA3 in TFT_SPI::analyzeText(char const*, unsigned short*, unsigned short*, unsigned short) at lib/tftLib/tft_spi.cpp:1476
  #5  0x42084BC2 in TFT_SPI::writeText(char const*, unsigned short, unsigned short, short, short, unsigned char, unsigned char, bool, bool, bool) at lib/tftLib/tft_spi.cpp:1795
  #6  0x42006F55 in textbox::writeText(char const*) at src/common.h:1851
  #7  0x420070C5 in showStreamTitle(char const*) at src/main.cpp:844
  #8  0x4202D105 in showStreamTitle(char const*) at src/main.cpp:835
      (inlined by) loop() at src/main.cpp:3072
  #9  0x420999BC in loopTask(void*) at C:/Users/Hoeckmayr/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:74

Es hilft nur ein Hardware-Reset oder Power-Off / On, die Schleife zu beenden. 
Ich versuche mal, ob der Fehler auch mit einem anderen Sender auftritt. 

Viele Grüße, 
Alfons

schreibfaul1, 2025-04-26:
Hi Alfons,
hab Arabella-Austropop laufen lassen und hatte nach kurzer Zeit denselben Fehler. Manchmal werden im "Streamtitle" Zeichen übertragen, die es (in der UTF-8 Tabelle) nicht gibt. Das führte in zu einer Endlosschleife und löste den Watchdog aus. Ich hoffe, das passiert jetzt nicht mehr.

beste Grüße
Wolle

TubaFonse, 2025-04-26:
Danke Wolle für die schnelle Hilfe!!! 
LG Alfons.

closed 2025-04-26 by TubaFonse