schreibfaul1/ESP32-MiniWebRadio #683: Audio Erweiterung WaveShare ESP32-S3 7"

Arne65, 2025-05-05:
Hallo Wolle 

ich habe mal eine PCB für das WaveShare ESP32-S3 7" Display (Audio) gemacht  @
2 X MAX98357A 3.2W Output Power into 4Ω at 5V ( ich betreibe es hier mit 3,3 V9 was voll ausreicht für einen Radiowecker etc.

1 X PCM5102 

[image]

[image]

eventuell kann jemand was damit anfangen 

werde das wieder auf OSWHLAB Veröffentlichen

Danke für deine ausgezeichnete Arbeit

schreibfaul1, 2025-05-06:
Hallo Arne,
super Sache! Ich glaube, die Waveshare Displays sind mehr gebräuchlich als von anderen Herstellern.
Wenn du magst, übernehme ich die Fotos in den Bereich der User-Geräte.

viele Grüße
Wolle

Arne65, 2025-05-06:
Hallo Wolle,

selbstverständlich kannst du das in den Bereich der User-Geräte übernehmen .

Mir gefiel das WaveShare Display am besten :) und die vier Befestigungen waren für mich der ideale Platz für die PCB, dabei bleibt das ganze  kompakt.

Finden kann man das Projekt unter:

 https://oshwlab.com/kubel/audio-display-waveshare

oder

https://easyeda.com/editor#project_id=98c890d47cb84ad787b6995b157c5ce4

Das kann aber noch ein paar Tage dauern, bis es freigeschaltet wird .

Nochmal vielen Dank für Deine hervorragende  Arbeit.

Einen kleinen Fehler hätte ich noch 

Beim Wecken werden nur die blinkenden Punkte angezeigt, aber nicht die aktuelle Uhrzeit.

Ansonsten  läuft das Radio perfekt auf dem Display, ich habe Version 1.2 vom Display, da hatte ich keine Probleme mit GND Verbindungen .

viel Grüße
 Arne

Arne65, 2025-05-08:
Hallo Wolle,

Danke sieht gut aus :)

viele Grüße

Arne

closed 2025-06-22 by github-actions[bot]