schreibfaul1/ESP32-MiniWebRadio #684: Normalizing (equalizing) the volume

valera39, 2025-05-06:
Hello Dear Wolle!
The fact is that radio stations broadcast at different volume levels. Therefore, when switching stations, you have to decrease or increase the volume for more comfortable listening. This is especially important for an alarm clock with a fixed wake-up volume. As a result, one station sounds loud and the other station sounds quiet.
Is it possible to equalize the initial volume level of each station, e.g. as it is done in Ka Radio?

schreibfaul1, 2025-05-07:
Hi Valera,
the problem is that I don't know the "amplitude swing" of the stations. Increasing the amplification for quiet passages will not be good. I have no idea about KARadio, does anyone know how this is done there?
Or you would have to set the volume for each station individually.

valera39, 2025-05-07:
If I understood how it worked in Ka ragio, I would certainly clarify. That's a pity, nevertheless thank you very much for your reply.
Let me ask you another question: is it possible to make a separate list of favorite stations? This wish has also been expressed by other users, e.g. https://github.com/clear-sky

schreibfaul1, 2025-05-07:
There's not much to see on KARadio's Github. I think a lot is stored in binary files.
That's different here, everything is openly accessible as source code.
To decide whether the level should be lowered or raised, the stream would have to run for some time.

Instead of favourites, I could insert a categorisation,
🟢🟡🔴🔵🟠🟣🟤🟩🟨🟥🟦🟧🟪🟫
then everyone can decide for themselves what a symbol stands for, e.g. oldies, pop etc

valera39, 2025-05-08:
A separate list on the display would be much more convenient, or is that not possible?

closed 2025-06-22 by github-actions[bot]