schreibfaul1/ESP32-MiniWebRadio #685: Feature Request: Touch screen calibration.

japywheel, 2025-05-09:
I use use a IL9341 2.4 inch screen (as 2.8 did not fit in enclosure) The touch calibration is way off. It would be nice to recalibrate the screen from the web interface.

Also is there a work around I have missed somewhere?

Thanks

omegaxxl, 2025-05-17:
Hi, I have the same problem. On the 2.8" display, the touch is also completely different. In common.h, I have the display and TP_VERSION set correctly. 
How to fix it?

TedWillemsen, 2025-06-06:
Hi **japywheel** and **omegaxxl**

Additional information:
I used the ILI9341(1) selection, (original mentioned for a 2.8" RPI version) because I also build a 2.8" TP version so I could easy switch
in programming between the 2 versions. The (not used) RPI version values are overwritten.
Of course it is possible to move the 2.4 " rework to the ILI9341(0) area but in that case the original 2.8" values are overwritten.

closed 2025-07-17 by japywheel