schreibfaul1/ESP32-MiniWebRadio #690: Problem after new install Linux Mint and VSCode

HWH53, 2025-05-28:
Hallo Wolle,
ich benötige wieder einmal deine Hilfe. Nach einem Hardwaredefekt meiner Festplatte musste ich Linux und alle weiteren Programme samt VSCode neu installieren. Natürlich waren auch die Timestampsicherungen auf einer Partition dieser Platte abgespeichert. Vorher funktionierte alles unter Windows wie geschmiert, nun bekomme ich dein geklontes Repositiory nicht mehr zum laufen.
Ich hänge die entsprechenden Logfiles an. 
Vielleicht hast du oder jemand aus der Community einen Tip für mich was ich falsch konfiguriert habe.

**Protokoll beim Clonen:**

`Resolving esp32s3 dependencies...
Platform Manager: Installing https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Platform Manager: espressif32@54.3.20 has been installed!
Tool Manager: Installing https://github.com/espressif/arduino-esp32/releases/download/3.2.0/esp32-3.2.0.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32@3.2.0 has been installed!
Tool Manager: Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.4/esp32-arduino-libs-idf-release_v5.4-2f7dcd86-v1.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32-libs@5.4.0+sha.2f7dcd862a has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp-idf/releases/download/v5.4.1/esp-idf-v5.4.1.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-espidf@3.50401.0 has been installed!
Tool Manager: Installing platformio/toolchain-xtensa-esp-elf @ 14.2.0+20241119
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing platformio/toolchain-riscv32-esp @ 14.2.0+20241119
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: toolchain-riscv32-esp@14.2.0+20241119 has been installed!
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.23800.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: toolchain-esp32ulp@1.23800.240113 has been installed!
Tool Manager: Installing platformio/tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-xtensa-esp-elf-gdb@14.2.0+20240403 has been installed!
Tool Manager: Installing platformio/tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-riscv32-esp-elf-gdb@14.2.0+20240403 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esptool/releases/download/v4.8.9/esptool.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esptoolpy@4.8.9 has been installed!
Tool Manager: Installing tasmota/tool-mklittlefs @ ^3.2.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70%
Unpacking 0% 10% 20%
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
Tool Manager: Installing platformio/tool-cmake @ ~3.30.2
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-cmake@3.30.2 has been installed!
Tool Manager: Installing platformio/tool-ninja @ ^1.7.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20%
Tool Manager: tool-ninja@1.7.1 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40801.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-scons@4.40801.0 has been installed!
Library Manager: Installing git+https://github.com/schreibfaul1/ESP32-audioI2S.git
git version 2.43.0
Klone nach '/home/hwh/.platformio/.cache/tmp/pkg-installing-csgjd8ts' …

Library Manager: ESP32-audioI2S@3.0.13+sha.870d48f has been installed!
Library Manager: Installing git+https://github.com/schreibfaul1/ESP32-KCX-BT-EMITTER.git
[... 295 lines cut ...]
    from kconfiglib.report import PRAGMA_PREFIX
  File "/home/hwh/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages/kconfiglib/report.py", line 29, in <module>
    from rich import print as rprint
ModuleNotFoundError: No module named 'rich'
CMake Error at /home/hwh/.platformio/packages/framework-espidf/tools/cmake/kconfig.cmake:209 (message):
  Failed to run kconfgen
  (/home/hwh/.platformio/penv/.espidf-5.4.1/bin/python;-m;kconfgen;--list-separator=semicolon;--kconfig;/home/hwh/.platformio/packages/framework-espidf/Kconfig;--sdkconfig-rename;/home/hwh/.platformio/packages/framework-espidf/sdkconfig.rename;--config;/home/hwh/Projects/Radio/ESP32-MiniWebRadio/sdkconfig.esp32s3_OTA;--env-file;/home/hwh/Projects/Radio/ESP32-MiniWebRadio/.pio/build/esp32s3_OTA/config.env).
  Error 1
Call Stack (most recent call first):
  /home/hwh/.platformio/packages/framework-espidf/tools/cmake/build.cmake:681 (__kconfig_generate_config)
  /home/hwh/.platformio/packages/framework-espidf/tools/cmake/project.cmake:717 (idf_build_process)
  CMakeLists.txt:3 (project)

========================================================================== [FAILED] Took 6.23 seconds ==========================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32s3_OTA    FAILED    00:00:06.225
===================================================================== 1 failed, 0 succeeded in 00:00:06.225 =====================================================================

 *  The terminal process "platformio 'run', '--environment', 'esp32s3_OTA'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
`

`

HWH53, 2025-05-28:
Hallo Wolle,

danke für die schnelle Antwort. Leider hilft mir dies nicht weiter, da ich bereits mehrfach VSCode + PlatformIO deinstalliert (alle Reste und Verzeichnisse manuell gelöscht habe) und dann VSCode neu installiert habe. Immer das gleiche Problem. Bin schier am verzweifeln weil ich nicht weiterkomme.

Immer der gleiche Fehler: "fatal: Kein Git-Repository (oder irgendeines der Elternverzeichnisse): .git". Das Verzeichnis .git ist im Dateiexplorer vorhanden, wird aber in VSCode nicht angezeigt. Ist dies korrekt?

git Status sagt ok. 
Der Fehler tritt bereits bei der Initialisierung auf. 

mfg

Helmut

schreibfaul1, 2025-05-30:
Das Git Log sieht bei mir ähnlich aus. Versuche in VSCode ein weiters Profil anzulegen.
Ich verwende Code nicht aus der Anwendungsverwaltung, sondern habe einen Eintrag in den Paketquellen.

[image]

HWH53, 2025-05-30:
Hallo Wolle,
danke für den Hinweis. 
Diese Microsoft Paketquelle ist bei mir ebenfalls eingetragen, da ich über ein, von der offiziellen Website, heruntergeladenes .deb File installiert habe. Ich wollte halt nicht aus der Anwendungsverwaltung installieren, da Code dort nur als Flatpak vorhanden ist.

Meine Überlegungen:
1.) VSCode Version fehlerhaft ???
`
Version: 1.100.2
Commit: 848b80aeb52026648a8ff9f7c45a9b0a80641e2e
Date: 2025-05-14T21:47:40.416Z
Electron: 34.5.1
ElectronBuildId: 11369351
Chromium: 132.0.6834.210
Node.js: 20.19.0
V8: 13.2.152.41-electron.0
OS: Linux x64 6.8.0-60-generic
`
2.) Git Version 2.43.0

Folgendes habe ich gestern getestet, da Code derzeit generell bei der Projektkonfiguration von V4 mit Fehler aussteigt:
Ich habe aus deinem Repository den Code für die v3.7 heruntergeladen, dieser wurde fehlerfrei in meiner VSCode-Installation initialisiert und lies sich ebenfalls problemlos compilieren. 
Wo liegen die Unterschiede zur V4?

Hast Du noch irgendeine Idee wer bei der Fehlsuche behilflich sein könnte?

Ich werde heute nochmals VSCode kompl. löschen und als Flatpak, in der Hoffnung das dort noch eine ältere VSCode Version enthalten ist, installieren um es dann damit nochmals zu testen. 

mfg
Helmut

closed 2025-05-30 by HWH53