schreibfaul1/ESP32-MiniWebRadio #693: Remote control settings

omegaxxl, 2025-05-30:
Hi,
I built this beautiful radio, but I have one problem. I can't set the codes from the remote control according to the instructions. The control is the same as the one on the left side of the picture in the instructions. After pressing the button, I click on the function I would like to control. The box turns yellow and the code 0x46 appears. The problem is that several other boxes change at the same time. They turn yellow and the same code 0x46 appears in them. The code is the same, no matter which button I press.
Can someone tell me where I am making a mistake and what to do to get it right?
Thank you for your help

[image]

schreibfaul1, 2025-05-30:
This looks strange. Each remote control has a fixed IR address. But each button has its own IR command.

0x46 cannot be used for several buttons at the same time. 

The remote control manufacturers use different commands, so you have to assign the buttons for your remote control once.  Press the "0" button on the remote control and then use the mouse to click on the "0" field. Do the same with button 1, 2, 3....

>

omegaxxl, 2025-05-30:
I tried it, but I had to clear all the codes. But still, sometimes the same code appears in multiple positions. I tried to make a tester to list the codes and they are fine. none are repeated.
Wouldn't it be better to have the option to enter the individual codes manually into a file?

schreibfaul1, 2025-05-30:
Yes, of course, in the SD root there is the file "ir_buttons.json"

````
[
  {
    "A": "0x01",
    "label": "IR address"
  },
  {
    "C": "0x13",
    "label": "IR command"
  },
  {
    "0": "0x01",
    "label": "ZERO"
  },
  {
    "10": "0x41",
    "label": "MUTE"
  },
  {
    "20": "0x40",
    "label": "ON/OFF"
  },
  {
    "30": "-1",
    "label": "CH+"
  },
  {
    "1": "0x4e",
    "label": "ONE"
  },
  {
    "11": "0x50",
    "label": "ARROW RIGHT"
  },
  {
    "21": "0x11",
    "label": "RADIO"
  },
  {
    "31": "-1",
    "label": "CH-"
  },
  {
    "2": "0x0d",
    "label": "TWO"
  },
  {
    "12": "0x51",
    "label": "ARROW LEFT"
  },
  {
    "22": "0x43",
    "label": "PLAYER"
  },
  {
    "32": "-1",
    "label": "-"
  },
  {
    "3": "0x0c",
    "label": "THREE"
  },
  {
    "13": "0x1a",
    "label": "ARROW DOWN"
  },
  {
    "23": "0x0f",
    "label": "DLNA"
  },
  {
    "4": "0x4a",
    "label": "FOUR"
  },
  {
    "14": "0x16",
    "label": "ARROW UP"
  },
  {
    "24": "0x19",
    "label": "CLOCK"
  },
  {
    "5": "0x09",
    "label": "FIVE"
  },
  {
    "15": "0x42",
    "label": "MODE"
  },
  {
    "25": "0x00",
    "label": "OFF TIMER"
  },
  {
    "6": "0x08",
    "label": "SIX"
  },
  {
    "16": "0x13",
    "label": "OK"
  },
  {
    "26": "0x18",
    "label": "VOLUME+"
  },
  {
    "7": "0x46",
    "label": "SEVEN"
  },
  {
    "17": "-1",
    "label": "-"
  },
  {
    "27": "0x10",
    "label": "VOLUME-"
  },
  {
    "8": "0x05",
    "label": "EIGHT"
  },
  {
    "18": "0x5a",
    "label": "PAUSE/RESUME"
  },
  {
    "28": "0x55",
    "label": "<< 30s"
  },
  {
    "9": "0x04",
    "label": "NINE"
  },
  {
    "19": "0x52",
    "label": "STOP"
  },
  {
    "29": "0x54",
    "label": ">> 30s"
  }
]

````

omegaxxl, 2025-05-30:
Sorry to take your time, but I'm a beginner. Could you please write my two codes as they should be? I hope I can understand it then.

schreibfaul1, 2025-05-30:
Each press of the button should display the corresponding command. If this does not happen, there may be a problem with the IR receiver

[image]

In the example, the address is 0x01 and the command 0x13

0x13 is inverted 0xEC

00010011 -> 11101100 

and what is sent is then 0113EC

omegaxxl, 2025-05-30:
I guess I'm done here. Every time I press a button, the number from the IR command appears in the window. The IR receiver is new and I tried another one, with the same result. I tried powering it with 3.3V and 5V and it's the same. I didn't understand the number conversion. The transmitter gives a four-digit number and I don't know how to shorten it. Maybe it's because I've been out of school for a long time (75+).

omegaxxl, 2025-05-30:
Looks like I won. Thank you very much for your patience. I was already thinking that all the work with the PCB design was useless.