schreibfaul1/ESP32-MiniWebRadio #696: Experimenting with Player and SYLT tag

slot00753, 2025-06-09:
[image]

I've taken the liberty of a little modification of the Player's look.
If someone is interested in using Synced Lyrics (SYLT) mp3 tag:
I used free SYLT editor 1.1.5 (http://compuphase.com), this free database: https://lrclib.net/
and code example from here: https://github.com/schreibfaul1/ESP32-audioI2S/blob/master/examples/Synchronised%20lyrics/synchronised_lyrics.ino
(modified, as the SYLT editor already gives big-Endian encoding)

schreibfaul1, 2025-06-09:
It turned out well and I like it. 
Kid3 can also be used as an editor.

[image]

schreibfaul1, 2025-07-17:
Hi Jan,
that's on the to-do list. But it's not that urgent because there are only a few files with SYLT (mp3 or flac is possible).

riks5, 2025-07-18:
Hi Wolle,
Thanks for your reply.
That would be great for an MP3!
Take your time, nothing urgent.
Best regards.
Jan

closed 2025-09-03 by github-actions[bot]