schreibfaul1/ESP32-MiniWebRadio #697: Kompilation Probleme

Robert-c-v, 2025-06-09:
Resolving esp32s3 dependencies...
Library Manager: Installing git+https://github.com/schreibfaul1/ESP32-audioI2S.git
git version 2.49.0.windows.1
Cloning into 'C:\Users\xxx\.platformio\.cache\tmp\pkg-installing-y8jbg8zr'...

Library Manager: ESP32-audioI2S@3.3.0+sha.49ccce0 has been installed!
Library Manager: Installing git+https://github.com/schreibfaul1/ESP32-KCX-BT-EMITTER.git
git version 2.49.0.windows.1
Cloning into 'C:\Users\xxx\.platformio\.cache\tmp\pkg-installing-83d6ykx9'...

Library Manager: ESP32-KCX-BT-EMITTER@0.0.0+20240128094011.sha.4c07f9b has been installed!
Library Manager: Installing git+https://github.com/schreibfaul1/ESP32-DLNA-Client.git#1e296d9
git version 2.49.0.windows.1
Cloning into 'C:\Users\xxx\.platformio\.cache\tmp\pkg-installing-90suzdgb'...

HEAD is now at 1e296d9 make udp more robust
Library Manager: ESP32-DLNA-Client@0.0.0+20231205185914.sha.1e296d9 has been installed!
Library Manager: Installing git+https://github.com/schreibfaul1/ESP32-IR-Remote-Control.git#c371281
git version 2.49.0.windows.1
Cloning into 'C:\Users\xxx\.platformio\.cache\tmp\pkg-installing-9je4dzvi'...

HEAD is now at c371281 new long pressed key impl.
Library Manager: ESP32-IR-Remote-Control@1.0.10+sha.c371281 has been installed!
Updating metadata for the vscode IDE...
UserSideException: Processing esp32s3 (board: ESP32-S3-DevKitC-1-N4R8; platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip; framework: arduino, espidf)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP32-S3-DevKitC-1-N4R8.html
PLATFORM: Espressif 32 (54.3.20) > Espressif ESP32-S3-DevKitC-1-N8R2 (8 MB QD FLASH, 2MB QD PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.0 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a 
 - framework-espidf @ 3.50401.0 (5.4.1) 
 - tool-cmake @ 3.30.2 
 - tool-esptoolpy @ 4.8.9 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.9.0 
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403 
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403 
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0) 
 - toolchain-riscv32-esp @ 14.2.0+20241119 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
*** "custom_component" is used to select managed idf components ***
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.49.0.windows.1")
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Alina/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/Alina/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/Alina/.platformio/packages/toolchain-xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Dependencies lock doesn't exist, solving dependencies.

.......................NOTICE: Updating lock file at C:\Users\xxx\Documents\ESP32-MiniWebRadio-ESP32-P4\dependencies.lock

Processing 11 dependencies:

[1/11] chmorgan/esp-libhelix-mp3 (1.0.3)

[2/11] espressif/cbor (0.6.0~1)

[3/11] espressif/esp-dsp (1.4.12)

[4/11] espressif/esp-sr (1.9.5)

[5/11] espressif/esp-zboss-lib (1.6.3)

[6/11] espressif/esp-zigbee-lib (1.6.3)

[7/11] espressif/esp_modem (1.4.0)

[8/11] espressif/mdns (1.8.2)

[9/11] espressif/network_provisioning (1.0.2)

[10/11] joltwallet/littlefs (1.20.0)

[11/11] idf (5.4.1)

-- Project sdkconfig file C:/Users/xxx/Documents/ESP32-MiniWebRadio-ESP32-P4/sdkconfig.esp32s3
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any of the parent directories): .git
Traceback (most recent call last):

  File "<frozen runpy>", line 198, in _run_module_as_main

  File "<frozen runpy>", line 88, in _run_code

  File "C:\Users\xxx\.platformio\penv\.espidf-5.4.1\Lib\site-packages\kconfgen\__main__.py", line 5, in <module>

    from .core import main

  File "C:\Users\xxx\.platformio\penv\.espidf-5.4.1\Lib\site-packages\kconfgen\core.py", line 29, in <module>

    import esp_idf_kconfig.gen_kconfig_doc as gen_kconfig_doc

  File "C:\Users\xxx\.platformio\penv\.espidf-5.4.1\Lib\site-packages\esp_idf_kconfig\gen_kconfig_doc.py", line 14, in <module>

    import kconfiglib.core as kconfiglib

  File "C:\Users\xxx\.platformio\penv\.espidf-5.4.1\Lib\site-packages\kconfiglib\__init__.py", line 3, in <module>

    from .core import Kconfig  # noqa F401

    ^^^^^^^^^^^^^^^^^^^^^^^^^

  File "C:\Users\xxx\.platformio\penv\.espidf-5.4.1\Lib\site-packages\kconfiglib\core.py", line 27, in <module>

    from kconfiglib.report import PRAGMA_PREFIX

  File "C:\Users\xxx\.platformio\penv\.espidf-5.4.1\Lib\site-packages\kconfiglib\report.py", line 29, in <module>

    from rich import print as rprint

ModuleNotFoundError: No module named 'rich'

CMake Error at C:/Users/Alina/.platformio/packages/framework-espidf/tools/cmake/kconfig.cmake:209 (message):
  Failed to run kconfgen
  (C:/Users/xxx/.platformio/penv/.espidf-5.4.1/Scripts/python.exe;-m;kconfgen;--list-separator=semicolon;--kconfig;C:/Users/xxx/.platformio/packages/framework-espidf/Kconfig;--sdkconfig-rename;C:/Users/Alina/.platformio/packages/framework-espidf/sdkconfig.rename;--config;C:/Users/Alina/Documents/ESP32-MiniWebRadio-ESP32-P4/sdkconfig.esp32s3;--env-file;C:/Users/xxx/Documents/ESP32-MiniWebRadio-ESP32-P4/.pio/build/esp32s3/config.env).
  Error 1
Call Stack (most recent call first):
  C:/Users/xxx/.platformio/packages/framework-espidf/tools/cmake/build.cmake:681 (__kconfig_generate_config)
  C:/Users/xxx/.platformio/packages/framework-espidf/tools/cmake/project.cmake:717 (idf_build_process)
  CMakeLists.txt:3 (project)

========================= [FAILED] Took 35.79 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
esp32s3        FAILED    00:00:35.787

schreibfaul1, 2025-06-10:
Kann sein, dass es existiert, aber in einer falschen virtuellen Umgebung

HWH53, 2025-06-13:
Hallo  Robert,
please check your **Python Default Compiler Path**. In my new installation it was missing and the .venv was not found automatically by vscode. Rich was not found but it was installed in .venv.

[image]

After adding this Path the compiler errors are gone and rich was found. 

Use this Platform 53.03.13, the others throw errors at me.

[image]

mfg
Helmut

schreibfaul1, 2025-06-13:
Probably "rich" is installed here:
`~/.platformio/penv/lib/python3.12/site-packages
`
and is missing here:
`~/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages`

try this 
`~/.platformio/penv/.espidf-5.4.1/bin/pip install rich`
or this
`~/.platformio/penv/.espidf-5.4.1/bin/python -m pip install rich`

this is for Linux, in other operating systems the .platformio folder may be somewhere else.

Robert-c-v, 2025-06-14:
Ich habe alles probiert , mögliche weise ich bin zu dumm.. bei   mir funktioniert nicht ...

ve2cfb, 2025-06-14:
Same here. 
No luck with the last Wolle suggestio.
I even tried a fresh install windows 11 on my LapTop and i get the same problem.
Couple days ago, i tried to install ESP-IDF on my desktop pc, and i had similer problems.
Is there some old versions that we can try beside Mini WebRadio.?

schreibfaul1, 2025-06-15:
I think the previous pioarduino version does without "rich".

    Arduino 3.1.3 IDF 5.3.2
    platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13-1/platform-espressif32.zip

I had the same problem, but was able to install Python-rich.

noanylag, 2025-06-15:
> Same here. No luck with the last Wolle suggestio. I even tried a fresh install windows 11 on my LapTop and i get the same problem. Couple days ago, i tried to install ESP-IDF on my desktop pc, and i had similer problems. Is there some old versions that we can try beside Mini WebRadio.?

Hello. I have the same problem and I don't know how to solve it. It's a big disappointment. Maybe you can and share the solution.

schreibfaul1, 2025-06-16:
This will probably be the error message:
`
  File "/home/wolle/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages/kconfiglib/report.py", line 29, in <module>
    from rich import print as rprint
ModuleNotFoundError: No module named 'rich'
CMake Error at /home/wolle/.platformio/packages/framework-espidf/tools/cmake/kconfig.cmake:209 (message)
`
This means that Python-rich is not found. The path may be different depending on your operating system.

It looks like this for me:

[image]

In my case, rich was installed here: ` ~/.platformio/penv/lib/python3.12/site-packages`
and was missing here: `~/.platformio/penv/.espidf-5.4.1/lib/python3.12/site-packages`

then install rich in your path:

[image]

I already have it

If you want to switch to this environment interactively:

`source ~/.platformio/penv/.espidf-5.4.1/bin/activate pip install rich`

Then go out again with `deactivate`.

Why this happens?
PlatformIO uses separate environments for certain frameworks (e.g. ESP-IDF) so that their Python dependencies remain separate. You may therefore have to install packages several times, depending on the context in which the script is running.

noanylag, 2025-06-17:
Wow! It worked! Thanks for the tip. Now everything compiles and I can continue and finish this wonderful project. Thank you!

ve2cfb, 2025-06-19:
Does not work on Windows.

noanylag, 2025-06-21:
> Не работает в Windows.

No, it works for me Win11.

ve2cfb, 2025-06-27:
Hello Wolle
deleted **.platformio**  according to your previous post but same problems.
So far, i re-installed everything, still cannot compile.
I got rid of "rich error" but having xtenza , git and cmake error.
By the way, last week i tried to install esp-idf with Platformio and ad similar problems.

Should i wait for a fix ? 

[Log 2025-06-27.txt](https://github.com/user-attachments/files/20951107/Log.2025-06-27.txt)

Thanks for your time.

schreibfaul1, 2025-06-27:
1) Open the bash terminal in VSCode.
2.  ```cd ~``` <Enter>
3. 
````
source ~/.platformio/penv/.espidf-5.4.1/bin/activate
pip install rich
````
<Enter>

[image]

ve2cfb, 2025-06-28:
Hello Wolle,

From in your previous post,  The RED 3 you have a bin directory but i dont have that.
You are probably on Linux and i am on Windows. 

The only place where i have "activate" is in he following directory.
Directory : C:\Users\Pierre\.platformio\penv\Scripts

=========================================================
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----        2025-06-28     05:47                __pycache__
-a----        2025-06-28     05:46           1768 activate
-a----        2025-06-28     05:46           1001 activate.bat
-a----        2025-06-28     05:46          26199 Activate.ps1
-a----        2025-06-28     05:46         108409 async-json-rpc-server.exe
-a----        2025-06-28     05:47           3856 bin2hex.py
-a----        2025-06-28     05:46         108392 bottle.exe
-a----        2025-06-28     05:46         180099 bottle.py
-a----        2025-06-28     05:46            393 deactivate.bat
-a----        2025-06-28     05:47           4715 hex2bin.py
-a----        2025-06-28     05:47           4580 hex2dump.py
-a----        2025-06-28     05:47           2888 hexdiff.py
-a----        2025-06-28     05:47           3700 hexinfo.py
-a----        2025-06-28     05:47           6245 hexmerge.py
-a----        2025-06-28     05:58         108407 markdown-it.exe
-a----        2025-06-28     05:46         108413 normalizer.exe
-a----        2025-06-28     05:46         108405 pio.exe
-a----        2025-06-28     05:46         108425 piodebuggdb.exe
-a----        2025-06-28     05:46         108410 pip.exe
-a----        2025-06-28     05:46         108410 pip3.11.exe
-a----        2025-06-28     05:46         108410 pip3.exe
-a----        2025-06-28     05:46         108405 platformio.exe
-a----        2025-06-28     05:58         108402 pygmentize.exe
-a----        2025-06-28     05:46         108407 pyserial-miniterm.exe
-a----        2025-06-28     05:46         108409 pyserial-ports.exe
-a----        2025-06-28     05:45         270616 python.exe
-a----        2025-06-28     05:45         259352 pythonw.exe
-a----        2025-06-28     05:46          92486 readelf.py
-a----        2025-06-28     05:58         108400 rich-click.exe
-a----        2025-06-28     05:46         108396 tabulate.exe
-a----        2025-06-28     05:46         108398 uvicorn.exe
-a----        2025-06-28     05:47         108395 wheel.exe
============================================================

My next step is going to Windows 11 and see what happen.

HWH53, 2025-06-28:
Hi VE2CFB,
i.ve looked at my working Win10 Installation and found the following in user/myhome/.platformio/penv/Scripts/.espidf-5.3.2/activate.bat

[image]

and then

[image]

i hope this helps a little bit.

mfg 
Helmut

ve2cfb, 2025-07-04:
Hello Wolle,

I installed Linux Mint, VS CODE, Platformio on a laptop and everything works fine.
Last time i used  Linux was about 50 + years ago and i only used it for a very short time. 
So now, i am in a learning curve.

I just add the rich stuff according to your post. 
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/697#issuecomment-2975172392

As far as Windows problem, is concerned, i will put it aside for a while.

Many thanks for your support.
Pierre

schreibfaul1, 2025-07-05:
If you don't need to use special software that only runs on Windows, Linux is a good choice.

closed 2025-08-20 by github-actions[bot]