schreibfaul1/ESP32-MiniWebRadio #703: MiniWebRadio Flac decoder

mguibord-prog, 2025-07-17:
Flac decoder seems to be crashing the sdcard and the system, will not reboot until I replace the sdcard. All other decoders work fine.
If I convert the the file from Flac to MP3 format it plays fine.

schreibfaul1, 2025-07-17:
There may be different values in the FLAC header. Do you have a sample file or URL?

mguibord-prog, 2025-07-18:
Here is a sample file link to download. 
https://drive.google.com/file/d/1ARDyd1CLEVs0TXg6fktDIlPc9DUfXqFK/view?usp=drive_link

This song will play for about 20sec and then crash the system and corrupt the sdcard.

TubaFonse, 2025-07-22:
I can play FLAC Files on SD-Card or NAS / DNS-Server without problems ... 
but when I try to play a Playlist .m3u which contents the url of FLAC-Files on my NAS / DNS-Server, the MWR crashes / reboots: 

e.g. content of my Playlist .m3u 

[image]

I hope this helps Wolle to reproduce the problem

greetings, 
Alfons.

TubaFonse, 2025-07-22:
Serial Monitor: 
23:19:05 webSrv: ...  Play "/audiofiles/Blechverrueckt_FLAC.m3u"
23:19:05 AUDIO_info:  buffers freed, free Heap: 133415 bytes
23:19:05 AUDIO_info:  connect to: "192.168.178.24" on port 50002 path "/transcoder/genericoder.cgi/id=32271.wav?format=WAV&downsample=true"
23:19:05 AUDIO_info:  Connection has been established in 13 ms
23:19:05 Playlist:    http://192.168.178.24:50002/transcoder/genericoder.cgi/id=32271.wav?format=WAV&downsample=true
23:19:06 ARDUINO:     volume old: 4. new: 4
23:19:07 webSrv: ...  Load from SD  "/stations.json"
23:19:08 webSrv: ...  Load from SD  "/ir_buttons.json"
23:19:08 sta 1, _cur_station 1
23:19:08 action: ...  switch to station 1

schreibfaul1, 2025-07-24:
@mguibord-prog has provided me with a flac file. When I play this from the SD card there are no problems. From the description I would assume that there is too little free memory, but this cannot be the case with the S3 with 4MB + PSRAM. It could also be due to the SD card or a poorly soldered connection.
I will continue testing.

TubaFonse, 2025-07-25:
I get the problems, only when I want to play the FLAC or M4A files in a PLAYLIST, which access the files from my DLNA-Server (NAS or PC or FritzBox)! 
Please try this test case to reproduce the FLAC problem: 
Copy the provided flac file on your DLNA-Server (PC or NAS), create with VLC Universal Plug'n'Play a playlist, put that playlist on SD card and play it in MWR Player mode.

TubaFonse, 2025-07-28:
Hallo Wolle, zunächst mal danke!!! 
Zur Info, verwirrt hat mich, dass nur die Playlists mit FLAC-Files's der Musikstücke - die ich wie die MP3's von der selben CD mit WindowsMediaPlayer rippte - nicht mit dem MWR Player abspielbar sind.  
Beim Erstellen der Playlist mit VLC werden die Umlaute in den TAGS meiner MP3-Dateien anders dargestellt als die Umlaute der TAGS von FLAC-Dateien. Ich habe gelesen: Umlaute in MP3-Tags sind kein Problem, wenn das verwendete System UTF-8 unterstützt. Bei älteren Systemen kann es zu Problemen kommen, die durch die Verwendung von UTF-8 oder durch die Aktualisierung der Software oder des Tagging-Formates behoben werden können. 
VLC speichert die Playlists im **ANSI** Code, die Umlaute werden unterschiedlich dargestellt, vermutlich je nachdem aus welcher Datenbank die Tags bezogen werden, siehe Vergleich (Hex) der Umlaute MP3 und FLAC in meiner Playlist: 

[image]

Ich habe auch in meiner NAS den Transcoder deaktiviert: 
[image]
wenn aktiviert, führt er auch zu einem langen String in der Playlist: 

[image]

 Grüße, Alfons

schreibfaul1, 2025-07-28:
Das sieht nach  ISO 8859-1 (Latin-1) aus.

[image]

Ich vermute, der VLC Player nimmt den Dateinamen wie er ihn in aus Quelle vorfindet. Dateinamen und Playlisten sollten immer im UTF-8 Format vorliegen. Der einfachste Weg ist die betreffenden Umlaute manuell  zu korrigieren. Wenn es viele Dateien sind und du auch die TAGs ändern möchtest geht das mit Kid3 ganz gut.

TubaFonse, 2025-07-29:
Danke, hab die TAGs in meinen FLAC-Dateien kurz manuell geändert. Schade, dass der WindowsMediaPlayer nach dem Einlesen der CD und erzeugen der FLAC-Datei nicht UTF-8 sondern die Windows Standard Zeichencodierung (ANSI) mit diesen komischen Umlauten/Sonderzeichen verwendet. Welcher Ripper kann von CDs direkt FLACs mit korrekten UTF-8 TAGs erzeugen? Beim WindowsMediaPlayer kann ich leider die Standard-Zeichenkodierung ANSI nicht verändern. Kid3 muss ich noch ausprobieren ...

closed 2025-09-13 by github-actions[bot]