schreibfaul1/ESP32-MiniWebRadio #705: AAC-64 Station: Software Reset due to exception/panic

TubaFonse, 2025-07-21:
Hallo Wolle, 
wenn ich folgenden Sender wähle (Radio Tirol): 

https://stream.radiotirol.it/live/aac-64/app/ 

folgt ein Software reset due to exception/panic, 
Ser.-Monitor: 

21:49:36 ir_code: ..  IR address 0x04, IR command 0x13
21:49:38 ir_result:   Stationnumber 3
00:00:00        *     MiniWebRadio V4     *    
00:00:00        ***************************    
00:00:00    
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ST7796
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 14746.9 MB by 15193.5 MB free
00:00:00 ARDUINO:     LEDC attached to pin 10 (channel 0, resolution 8)
00:00:00 ARDUINO:     [WIFI][APlistAdd] add SSID: FRITZ!Box7490LF
00:00:00 WiFI_info:   add credentials: FRITZ!Box7490LF - xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx main.cpp:1294]
00:00:00 ARDUINO:     [WIFI][APlistAdd] add SSID: Nw5Box7490GastLF
00:00:00 WiFI_info:   add credentials: Nw5Box7490GastLF -  [main.cpp:1294]
00:00:00 ARDUINO:     [WIFI] scan done
00:00:00 ARDUINO:     [WIFI] 4 networks found
00:00:00 ARDUINO:     [WIFI] Connecting BSSID: C8:0E:14:0C:93:11 SSID: FRITZ!Box7490LF Channel: 11 (-50)
00:00:00 ARDUINO:     STA already disconnected.
00:00:00 ARDUINO:     [WIFI] Connecting done.
00:00:00 WiFI_info:   Connecting WiFi...
00:00:00 WiFI_info:   mDNS name: MiniWebRadio
00:00:00 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.25, Received Signal Strength -48 dB
00:00:00 ARDUINO:     OTA server at: MiniWebRadio.local:3232
00:00:00 ftpServer:   Buffers allocated: 4622 bytes
00:00:00 rtime_info:  Initializing SNTP
00:00:00 AUDIO_info:  audioI2S Version 3.4.0a
00:00:00 AUDIO_info:  commFMT = Philips
00:00:00 setup: ....  number of saved stations: 195
00:00:00 setup: ....  number of saved favourites: 194
00:00:00 setup: ....  current station number: 3
00:00:00 setup: ....  current volume: 4
00:00:00 setup: ....  volume steps: 21
00:00:00 setup: ....  volume after alarm: 12
00:00:00 setup: ....  last connected host: https://stream.radiotirol.it/live/aac-64/app/
00:00:00 setup: ....  connection timeout: 2500 ms
00:00:00 setup: ....  connection timeout SSL: 3500 ms
00:00:00 RESET_REASON:Software reset due to exception/panic
00:00:00 Country: ... IT
21:49:52 WiFI_info:   RSSI is -47 dB
21:49:53 ftpServer:   Ftp server waiting for connection on port 21
21:49:53 BT-Emitter:  KCX_BT_Emitter found 
21:49:54 webSrv: ...  Load from SD  "/ir_buttons.json"
21:49:54 BT-Emitter:  Status -> Scan...
21:49:54 webSrv: ...  Load from SD  "/stations.json"
21:49:55 BT-Emitter:  Status -> Disconnected
21:49:55 BT-Emitter:  POWER ON 
21:49:55 BT-Emitter:  POWER ON 
21:49:55 BT-Emitter:  scanned: MacAdd:449f4910164a,Name:WZ-BT5.0
21:49:55 BT-Emitter:  CON LAST 
21:49:55 BT-Emitter:  Version KCX_BT_RTX_V1.4
21:49:55 BT-Emitter:  Volume -> 31
21:49:56 BT-Emitter:  Status -> Connected
21:50:01 DLNA_server: [0] 192.168.178.1:49000  FRITZ7490
21:50:01 DLNA_server: [1] 192.168.178.24:50001  DS213J_AL
21:50:01 DLNA_server: 2 media server found

Bitte um Hilfe, 

Danke&Grüße, 
Alfons

schreibfaul1, 2025-07-22:
Die selbst eingebauten Fehler sind am schwierigsten zu finden.
https://stream.radiotirol.it/live/aac-64/app/ leitet weiter zu einer URL die Prozentzeichen beinhaltet.

Was immer geht: printf("Hallo");
was zum Absturz führt: printf("Hallo%");

richtig wäre printf("Hallo%%"); oder printf("%s", "Hallo%");

habs korrigiert.

TubaFonse, 2025-07-22:
Danke Wolle für die anstrengende "Nachtschicht",
Grüße, Alfons

closed 2025-07-22 by TubaFonse