schreibfaul1/ESP32-MiniWebRadio #718: Audio Problem with I2S

ve2cfb, 2025-09-05:
Hello Wolle,

for the last 2 weeks, may radio stop playing with the VuMeter  in stop mode. The radio still work as far as accessing the different menu.
By going to the file list, i am able to pick the same station and every thing works ok.

Here is a log when the error happen. By the way, its always the same error message.

[image]

Thanks for your time.  Pierre

schreibfaul1, 2025-09-05:
Hello Pierre,
There are two I2C devices that can be connected: the brightness sensor and a capacitive touchpad.
If you don't have either of these, you can set SDA and SCL to -1, which will keep I2C inactive.
Otherwise, it is usually a wiring error, with SDA and SCL swapped or a connection interrupted.

ve2cfb, 2025-09-06:
Hello Wolle,
Thanks for the reply.
I was already aware of the SDA and SCl to -1. 

I will double check my connections just in case.

Like i said in my previous post, it usually happen a lot more early in the morning and  a lot less later in the afternoon .
This problem has been constant for the last 4 weeks. I know it does not make sense.
Initially, i tough that i had Network Problems. It`s happening a lot less with North American  Stations. 

Pierre

ve2cfb, 2025-09-13:
Hello Wolle,

It seems that **Version 4.0-rc3d   - Sep 09/2025** has fixed the problem.
For the last 3 days, no problems at all. 

Many Thanks...